realsenseai/librealsense #3270: Add minor version to SONAME

morxa, 2019-02-16:
Reading the discussion in #2994, you seem to only bump the minor version if you add incompatible API/ABI changes. Although this is somewhat uncommon, it's a perfectly viable option.

However, in that case, I would suggest that you should also add the minor version to the soname. The purpose of the soname is to detect that a library is binary-compatible, i.e., you should be able to replace any older version of the library with a newer version with the same soname. This assumption does not hold if you remove functions without bumping the soname. To quote [Wikipedia](https://en.wikipedia.org/wiki/Soname):
> If the application binary interface (ABI) of a library changes in a backward-incompatible way, the soname would be incremented, e.g. from libX.so.5 to libX.so.6.

Using the same soname for incompatible versions defeats the purpose of the soname.

Context: I'm the Fedora packager of librealsense. Currently, I have to manually check every update for ABI changes. Even worse, the dynamic linker would use a newer version of the library to satisfy the dependency on an older version of the library, because they share the same soname, even though that version is not compatible. This results in undefined symbols errors by the linker. Fedora (and I assume other distros, too) use the soname to track dependencies to inform package maintainers if a a rebuild is necessary. This mechanism is broken by using the same soname for incompatible versions.

I'd be happy to file a PR if you agree that the minor version should be added to the soname.

dorodnic, 2019-02-16:
Hi @morxa 
Thank you for the feedback. I'd like to discuss with my colleagues and will update this thread.

ev-mp, 2019-02-18:
@morxa, 
We've discussed this internally, and while we may disagree on the premise, including MINOR in soname will add to SDK's flexibility.
To be specific - it will allow for several SDK **runtime** environments to be installed and invoked simultaneously. (Note that the dev-package will remain single)

We're adding this request to the backlog. Nonetheless -  you're welcomed to submit PR on your own to expedite this change

dorodnic, 2019-03-05:
This seem to be slipping from the 2.19 release, but we should definitely implement it for the next release.

closed 2019-03-07 by RealSenseCustomerSupport

morxa, 2019-03-10:
I think this was closed on accident, 2.19.1 still doesn't have the minor version in the soname.

PR #3449 (2019-03-25): Add minor version to soname
On incomatible changes, only the minor version is bumped. Therefore, it
needs to be part of the soname to avoid incompatible library versions
with the same soname.

This resolves #3270.