realsenseai/librealsense #3271: rs2::points::get_texture_coordinates() explanation

spaltiel, 2019-02-16:
Camera Model: D435 
Firmware Version: 5.11
Operating System & Version: Win 10 
Platform: PC
SDK Version: 2.17
Language: C#/C++ 

Trying to understand what get_texture_coordinates() returns not from SW point of view but the math and physics?
Is it pixel location after projecting the XYZ points to the sensor?
We would like to convert each 3D point to two angles.

Can you please add explanation for each member/method in the API? 
for example see:
https://intelrealsense.github.io/librealsense/doxygen/classrs2_1_1points.html#a78d137348fc84b97a4057bd7c448bb2b
or 
https://intelrealsense.github.io/librealsense/doxygen/structrs2_1_1texture__coordinate.html
How can I figure without opening an issue what are u & v?

Best Regards,
Amir Itah

lramati, 2019-02-16:
In 3D computer graphics applications, texture coordinates indicate where to
sample the color image for the color of the point. So for instance, if the
point x=0,y=0,z=0 had texture coordinates u=5,v=4, you would color the
origin with the color of the (5,4) pixel of the color frame. Fractional
texture coordinates are usually interpreted as an interpolation of adjacent
pixels.

spaltiel, 2019-02-16:
Thanks! 
rs2::texture_coordinate indeed enable us to match each 3D point to the relevant index in the 2D image. We able to get cloud and and match the RGB data. 
Is there any explanation for the calculation of get_texture_coordinates()?

BTW,
Is it possible to add explanation for each member/method/... in the API?

MartyG-RealSense, 2019-02-16:
The developer UnaNancyOwen has an excellent SDK documentation site that can be navigated with menu selections at the top of the page.  The information is reliable because it is drawn directly from the official SDK documentation data and rearranged into a more user-friendly format.

https://unanancyowen.github.io/librealsense2_apireference/

Intel do not plan to add documentation for most of the Advanced Mode variables though.  This is because the complex way in which they interact with each other means that Intel prefer to handle them with machine learning.  You are free to experiment with the settings with trial-and-error though.

spaltiel, 2019-02-16:
10X!

closed 2019-02-16 by spaltiel

sandilyasg, 2023-03-10:
> 

@AItah @MartyG-RealSense 
Suppose I have  3D point cloud in the form of sensor_msgs::PointCloud2 and I want to match each 3D point to the relevant index in the 2D RGB image, how would I do this? Will I need to use get_texture_coordinates() or is there a better way? Essentially, I need to find the points from a RGB frame that correspond to the 3D point cloud generated by realsense-ros with the L515.