realsenseai/librealsense #3272: Recommended resolution support for D435 on USB 2.0 connections

ban1080, 2019-02-16:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   | (05.11.01.00) | 
| Operating System & Version |   {Rasbian} | 
| Kernel Version (Linux Only)    |  (4.16.??)                                         | 
| Platform                                 | Raspberry Pi |
| SDK Version                            |                           | 
| Language                            |  {C}                          | 
| Segment			|  {others }                          | 

### Issue Description
I realise that it is not possible to use the D435 with the recommended resolution of 848x480 on USB 2.0 systems.

Is it possible to add support for that as bandwidth is not an issue considering there is 1280x720 support on USB 2.0.

MartyG-RealSense, 2019-02-16:
This is not an official answer as I do not work for Intel.  It is unlikely that another mode will be added to USB2 due to Intel wanting to keep in check the number of modes they support (partly because more modes means extra testing).  For example, it was recently stated that there are no plans to support 24 FPS.  

In September 2018's RealSense online seminar, it was mentioned that a new 'Modified Huffman' lossless compression will enable the streaming of higher resolution and FPS on USB2, and more channels in multi-camera configurations.

RealSenseCustomerSupport, 2019-02-20:
----------------------------------------------

I submitted this change request to development team. They will handle it in lower priority.
Actually for D435, 848x480 provides better depth.

Thanks ban1080 for the suggestion.

ban1080, 2019-02-21:
Thank you.

Yes, We'd like for 848x480 to be available on USB 2.0.

closed 2019-02-21 by ban1080