realsenseai/librealsense #3273: Reducing pointcloud depth in Unity wrapper's point cloud example

akashcastelino, 2019-02-17:
Hi,

I am using the D435 camera in Unity using the Unity wrapper's point cloud color and depth example.

It renders the entire point cloud visible till the camera's 10m range. I was wondering if there is a way to reduce the point cloud to only a range between say 3m to 7m and as such remove the remaining parts of the point cloud?

Thanks in advance

MartyG-RealSense, 2019-02-17:
The ideal approach would be to use a 'depth clamp' function, which allows you to define a maximum distance (MaxZ), beyond which data past that point is ignored.

Unfortunately, most examples of that instruction are in C++ or Python language, rather than the C# code that the Unity wrapper uses.

https://github.com/IntelRealSense/librealsense/issues/2199#issuecomment-410683830

Another approach would be to reduce how far the camera can see by using a Disparity Shift.  As the value of the Disparity Shift increases, the maximum distance that the camera can see reduces.  I do not know how to set that in Unity either though.  Hopefully, one of the Intel staff on this forum can offer suggestions about how to do a depth clamp or disparity shift in C# in Unity.

A third option, that may be easier to implement, may be to set up a camera configuration file known as a JSON, and load that into your project to reconfigure the camera settings, defining the max depth-clamp distance within that file.

[image]

You can view some example json files here:

https://github.com/IntelRealSense/librealsense/wiki/D400-Series-Visual-Presets#preset-table

ogoshen, 2019-02-19:
In Unity there's a custom processing-block (unoptimized) implementation that does depth cutoff.
Source is [here](https://github.com/IntelRealSense/librealsense/blob/development/wrappers/unity/Assets/RealSenseSDK2.0/Scripts/ProcessingBlocks/DepthCutoff.cs). It's missing the near cutoff but that should be easy to add.

[image]

There's also a new native `threshold` processing-block that does exactly that, but didn't make it's way yet to c# & Unity... keep tuned though

akashcastelino, 2019-02-21:
Hi @ogoshen , I tried changing the values of depth cutoff's distance field, but the depth of the point cloud is not reducing.

[image]
[image]

ogoshen, 2019-02-24:
@akashcastelino you need to reorder the blocks so that the cutoff comes before the point cloud.

akashcastelino, 2019-02-25:
@ogoshen cool that worked ! Thank you !

closed 2019-02-25 by akashcastelino