realsenseai/librealsense #3274: 3d coordinates

picturedeeplover, 2019-02-18:
After using "rs2_deproject_pixel_to_point(upoint, &intr, upixel, udist)" to obtain 3d coordinates, x and z are very accurate, y error is more than ten centimeters, who can help me, thank you

dorodnic, 2019-02-18:
Hi @picturedeeplover 
What camera and version of software are you using?

picturedeeplover, 2019-02-19:
I use SR300 and SDK2.0, and use the function of Intel RealSense SDK2.0 \samples\measure.

RealSenseCustomerSupport, 2019-02-21:
----------------------------------------------

@picturedeeplover Which librealsense version are you using to see this issue?

RealSenseCustomerSupport, 2019-02-26:
----------------------------------------------

@picturedeeplover One more question - how did you measure when get the issue? Could you please provide some snapshots of your testing?

RealSenseCustomerSupport, 2019-02-28:
----------------------------------------------

@picturedeeplover Are you still having the issue? Please provide more information so that we can further check. Thanks!

picturedeeplover, 2019-02-28:
Suppose I take the infrared camera as the origin, fix the camera, then put an object in front of the camera, and output the coordinates of a point, compared with my actual measurement, the result shows that the z direction is very accurate, and there is a big error in the x or y direction. About coordinates, I want someone to tell me what the origin is, the x and y direction, thank you.

picturedeeplover, 2019-02-28:
Could you please provide the email address? I can show you the pictures.

RealSenseCustomerSupport, 2019-02-28:
----------------------------------------------

@picturedeeplove  You can attach the picture here.

RealSenseCustomerSupport, 2019-02-28:
----------------------------------------------

@picturedeeplover How was the camera fixed? Is it possible that there're some angles to cause the error?

picturedeeplover, 2019-03-02:
The camera is simply horizontal and facing the target, and there may be a small error, but the error is far less than that. As for pictures, I failed to add them in the comments section.

RealSenseCustomerSupport, 2019-03-04:
----------------------------------------------

@picturedeeplover Which librealsense version do you use?  Is the picture very large? Maybe you can compress it to try again. Thanks!

picturedeeplover, 2019-03-05:
I've got a problem! Depth maps that are not aligned with the color map are accurate in 3d coordinates. Once aligned, the resolution becomes 1920*1080, and the 3d coordinates are inaccurate. Is there a solution? thank you

RealSenseCustomerSupport, 2019-03-08:
----------------------------------------------

@picturedeeplover What's the resolution you configured for depth and color before align?

picturedeeplover, 2019-03-10:
The default：Color 1920*1080 depth 640*480

RealSenseCustomerSupport, 2019-04-26:
----------------------------------------------

If you need you can try to align to depth.
rs2::align align_to_depth(RS2_STREAM_DEPTH);

When you align to color, the depth image will be modified. but the intrinsics of depth isn't changed.

RealSenseCustomerSupport, 2019-05-05:
----------------------------------------------

@picturedeeplover Did you get chance to try align to depth? Looking forward to your update. Thanks!

RealSenseCustomerSupport, 2019-05-13:
----------------------------------------------

@picturedeeplover Any update? Thanks!

picturedeeplover, 2019-05-28:
Sorry, I am busy recently. Do you mean to align the color map with the depth map and reduce the resolution to 640*480? I have another problem, that is, the image rendered by the D435i camera using the color-map function cannot be displayed with Mat. Thank you

RealSenseCustomerSupport, 2019-08-26:
----------------------------------------------

@picturedeeplover Did you see the issue if align to depth? Please create another ticket for the image display issue. Thanks!

RealSenseCustomerSupport, 2019-09-09:
----------------------------------------------

@picturedeeplover Any other questions about this ticket? Thanks!

RealSenseCustomerSupport, 2019-09-16:
----------------------------------------------

@picturedeeplover If no other questions, will close the ticket accordingly. Thanks!

closed 2019-09-16 by RealSenseCustomerSupport