realsenseai/librealsense #3275: Impossible to load json file c#

Macxou, 2019-02-18:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415                                                        | 
| Firmware Version                   |  2.17.0                                                     | 
| Operating System & Version |   {Win (10)   | 
| Platform                                 | PC                                                          |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |       C#                                                       | 
| Segment			|                                                                       | 

----------------------------------------------------------------------------------------------------

### Issue Description
I want to load some json file load into the code from System.IO.StreamReader.
I have found how to access the Json configuration

`Device dev= contex.QueryDevices(false)[0]`
` adv = AdvancedDevice.FromDevice(dev);`
`adv.JsonConfiguration = loadedConf.Replace("\r\n", "\n");`

But when i use this setter, it doesn't change the configuration. 
It's possible that I do something wrong. Could you help me please ?

dorodnic, 2019-02-18:
It really depends on the rest of the code.
If you are using the pipeline, try setting JSON on the device you get from pipeline:
```cs
Pipeline p;
PipelineProfile f = p.Start();
adv = AdvancedDevice.FromDevice(f.Device);
adv.JsonConfiguration = loadedConf.Replace("\r\n", "\n");
```
Something like this, I think it should work.

Macxou, 2019-02-18:
Thank you for the answer. 
Now, I can set the JsonConfiguration. 
There is just one more thing, after loading the Json configuration from a file I get a System.AccessViolationException in the block process when it gets to the creation of the pointCloud.

                        `dev.block.Start(f =>
			{
				try
				{

					using (var frames = FrameSet.FromFrame(f))
					{

						dev.colorFrame = frames.ColorFrame.DisposeWith(frames);
						dev.depthFrame = frames.DepthFrame.DisposeWith(frames);

						pointCloud.MapTexture(dev.colorFrame);
						var pointTemp = pointCloud.Process(dev.depthFrame) as Points;
						if (pointTemp != null)
							dev.points = pointTemp;

						UpdateImage(dev);
						//dev.rdy = true;

					}
				}
				catch (Exception ex)
				{
					//MessageBox.Show(ex.Message + "\n\nstart");
					dev.rdy = true;
				}`

I have set up the pipeline like that : 
                                               `Pipeline pipeline = new Pipeline();
						string serial = dev.Info[CameraInfo.SerialNumber];
						Config cfg = new Config();
						cfg.EnableStream(Stream.Depth, width, height, Format.Z16, 6);
						cfg.EnableStream(Stream.Color, width, height, Format.Rgb8, 6);
						cfg.EnableDevice(serial);

						PipelineProfile profile = pipeline.Start(cfg);`
where dev = Device dev, and the pointCloud is initialized like that : PointCloud = new PointCloud();
Do you have any idea where it can come from ? Please
Regards

ogoshen, 2019-02-19:
Probably related to #3250

@Macxou try hanging on to the `AdvancedDevice` instance to keep it from being garbage collected.

Macxou, 2019-02-19:
Thank you for your advice !

RealSenseCustomerSupport, 2019-02-25:
----------------------------------------------

Hi Macxou,

Glad to hear that dorodnic & ogoshen's suggestions addressed your issues.
If there is nothing else needed, this one will be closed soon.

Thanks!

closed 2019-03-14 by RealSenseCustomerSupport