realsenseai/librealsense #3276: Depth and color frames have different resolutions

EmmanuelCos, 2019-02-18:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Operating System & Version |   Win 10| 
| Language                            |  C++            | 

Hello,

I am trying to save depth and color image separately. To do so I first get the depth and color frame, and retrieve their intrinsic information (depth, width, bit per sample...)

			rs2::pipeline p;
			rs2::config c;
			c.enable_device(serial_number);
			// Start the pipeline with the configuration
			rs2::pipeline_profile profile = p.start(c);
			auto frames = p.wait_for_frames();
			auto depth = frames.get_depth_frame();
			auto color = frames.get_color_frame();
			int widthC = color.get_width();
			int heightC = color.get_width();
			int widthD = depth .get_width();
			int heightD = depth.get_width();

I noticed that the depth resolution is 1280 * 720, which seems normal and coherent with what we get with realsense viewer, but the color one is 640 * 480. How could I do so that depth and color have the same resolution? Could I have the color frame in 1280 * 720?

Thanks and regards,

Emmanuel

dorodnic, 2019-02-18:
You can select resolution via `c.enable_stream(RS2_STREAM_COLOR, w, h, RS2_FORMAT_RGB8, 30)`

EmmanuelCos, 2019-02-18:
Ok thank you. Is it the instrinsit resolution of the RGB camera, or in this case the image would have been resized by the lib?
Thank you,
Emmanuel

dorodnic, 2019-02-18:
This would re-configure the hardware (not a software resize)

closed 2019-03-07 by RealSenseCustomerSupport