realsenseai/librealsense #3277: How to load a JSON file in a C++ script, for use in a Processing sketch

lvitters, 2019-02-18:
| Camera Model | D435 |
| Firmware Version | 05.10.13.00 |
| Operating System & Version | Win 10 |
| Platform | PC |
| SDK Version | 2.17.0 |
| Language | C++ |
| Segment | others |

I am trying to load a JSON setup file that I have created with the RealSense Viewer from a C++ script . I am not actually planning to use C++ to do something with the camera, but rather to open this short program from a Processing sketch, as unfortunately, there is no Java wrapper with this functionality. I have both successfully opened this program from my Processing sketch and received the depth stream in it, but it does not load the JSON, leaving me with an "un-"configured stream. 

Is what I am trying to do theoretically possible and if so, why is my C++ program not applying the JSON file?

`#include <iostream>
#include <fstream>
#include <string>

#include <librealsense2/rs.hpp>

#include <librealsense2/rs_advanced_mode.hpp>

using namespace rs2;
using namespace std;

void main()
{
	rs2::context ctx;
	auto devices = ctx.query_devices(); 

	for (int i = 0; i < devices.size(); i++)
	{	
		//get device
		rs2::device dev = devices[i];
		std::string dev_serial_number(dev.get_info(RS2_CAMERA_INFO_SERIAL_NUMBER));
		cout << "Device found" << endl;

		//Get the advanced mode functionality
		auto advanced_mode_dev = dev.as<rs400::advanced_mode>();

		//Load and configure .json file to device
		ifstream t("C:/Users/Lucca/Desktop/git/tests/realsense_library_test/reset/goodSettings.json");
		string str((istreambuf_iterator<char>(t)), istreambuf_iterator<char>());

		advanced_mode_dev.load_json(str);

		cout << "loaded" << endl;
	}
}`

AntoineWefit, 2019-02-18:
I don't really know what you are missing, but I know this bit is working for me : 
```
		auto device = ctx.query_devices();
		auto dev = device[0];

		if (strcmp(dev.get_info(RS2_CAMERA_INFO_NAME), "Intel RealSense D415") == 0) // Check for compatibility, must have if executed on a computer
		{
			config cfg;
			string serial = dev.get_info(RS2_CAMERA_INFO_SERIAL_NUMBER);
			string json_file_name = "17.json";
			cout << "Configuring camera : " << serial << endl;

			auto advanced_mode_dev = dev.as<rs400::advanced_mode>();

			// Check if advanced-mode is enabled to pass the custom config
			if (!advanced_mode_dev.is_enabled())
			{
				// If not, enable advanced-mode
				advanced_mode_dev.toggle_advanced_mode(true);
				cout << "Advanced mode enabled. " << endl;
			}

			// Select the custom configuration file
			std::ifstream t(json_file_name);
			std::string preset_json((std::istreambuf_iterator<char>(t)), std::istreambuf_iterator<char>());
			advanced_mode_dev.load_json(preset_json);
			cfg.enable_device(serial);
		}
		else
		{
			cout << "Selected device is not an Intel RealSense D415, check the devices list. " << endl;
		}
```
I deleted a bunch of lines and it's for a D415 but you get the idea...
Hope this'll help !

lvitters, 2019-02-18:
Thanks for your help. How do you display the camera feed after applying the setup? I think that is where my problem is, and I can not find any documentation about that online. Right now I am doing it like this, but I don't think that is right:

````C++
    // Declare RealSense pipeline, encapsulating the actual device and sensors
    rs2::pipeline pipe;

    // Start streaming with default recommended configuration
    pipe.start();

	using namespace cv;
    const auto window_name = "Display Image";
    namedWindow(window_name, WINDOW_AUTOSIZE);

    while (waitKey(1) < 0 && cvGetWindowHandle(window_name))
    {
        rs2::frameset data = pipe.wait_for_frames(); // Wait for next set of frames from the camera
		rs2::frame depth = data.get_depth_frame(); // .apply_filter(color_map);

        // Query frame size (width and height)
        const int w = depth.as<rs2::video_frame>().get_width();
        const int h = depth.as<rs2::video_frame>().get_height();

        // Create OpenCV matrix of size (w,h) from the colorized depth data
        Mat image(Size(w, h), CV_8UC3, (void*)depth.get_data(), Mat::AUTO_STEP);

        // Update the window with new data
        imshow(window_name, image);
    }
````

AntoineWefit, 2019-02-18:
Unfortunately, I'm far from being an expert and my personal application doesn't require any live streaming so I don't have the answer for that question. However, you can find some examples [here](https://github.com/IntelRealSense/librealsense/tree/master/examples) which might came handy !

Also, I'm not sure if I understood what really is your problem : are you able to display something but its "un-configured" or you can't display anything at all ?

lvitters, 2019-02-18:
Actually, I am abled to display something, but it's "un-configured" (it does not look like when I saved the JSON in the Viewer). 
My guess is that it actually is configuring the camera correctly, but they way I display the stream overrides that configuration. Unfortunately, Intel is not providing a minimal example of how to display the camera's depth stream as-is, all examples contain some form of configuration.

My end goal here is to configure the camera using a C++ program, but then receive the actual (configured) depth stream in another program (in my case a Processing sketch, which is Java). 

I am wondering if, in general, configuring the camera with a JSON is a system wide thing, or if it only "counts" for the particular program I do it in.

AntoineWefit, 2019-02-18:
> I am wondering if, in general, configuring the camera with a JSON is a system wide thing, or if it only "counts" for the particular program I do it in.

The settings are indeed, not saved "in" the camera, I think it comes with defaults settings which you can replace with a ".json".

Did you take a look at [this](https://github.com/IntelRealSense/librealsense/tree/master/examples/capture) ? I think it's pretty close to what you want to do.

AntoineWefit, 2019-02-18:
Here you are NOT applying anything to the pipeline :) : 
```
// Start streaming with default recommended configuration
pipe.start();
```

Try : 
```
pipeline pipe;
config cfg;
string serial = dev.get_info(RS2_CAMERA_INFO_SERIAL_NUMBER);
string json_file_name = ""C:/Users/Lucca/Desktop/git/tests/realsense_library_test/reset/goodSettings.json"";

cout << "Configuring camera : " << serial << endl;

auto advanced_mode_dev = dev.as<rs400::advanced_mode>();

// Check if advanced-mode is enabled to pass the custom config
if (!advanced_mode_dev.is_enabled())
	{
		// If not, enable advanced-mode
		advanced_mode_dev.toggle_advanced_mode(true);
		cout << "Advanced mode enabled. " << endl;
	}

std::ifstream t(json_file_name);
std::string preset_json((std::istreambuf_iterator<char>(t)), std::istreambuf_iterator<char>());
advanced_mode_dev.load_json(preset_json);
cfg.enable_device(serial);
pipe.start(cfg);
```
(Maybe not perfect but it should be close to work)

Hope this'll help !

lvitters, 2019-02-18:
The Java wrapper I am using has to start a pipe internally as well, which would (as I had guessed) override the cfg passed to the pipe here. Nonetheless, thank you for your help!

dorodnic, 2019-02-18:
Hi @lvitters 
Could you please share the JSON? 
Loading a JSON should be "sticky", meaning it should be preserved between application runs until you disconnect the device. Also, `pipeline.start` should not override any configuration.
This is with one exception of streaming parameters, such as width, height and fps. 
These are pretty much ignored by the library and handled at application level, for instance the realsense viewer. This is why I'd like to get the sense of what you are trying to load.

lvitters, 2019-02-19:
````JSON
{
    "aux-param-autoexposure-setpoint": "1536",
    "aux-param-colorcorrection1": "0.298828",
    "aux-param-colorcorrection10": "0",
    "aux-param-colorcorrection11": "0",
    "aux-param-colorcorrection12": "0",
    "aux-param-colorcorrection2": "0.293945",
    "aux-param-colorcorrection3": "0.293945",
    "aux-param-colorcorrection4": "0.114258",
    "aux-param-colorcorrection5": "0",
    "aux-param-colorcorrection6": "0",
    "aux-param-colorcorrection7": "0",
    "aux-param-colorcorrection8": "0",
    "aux-param-colorcorrection9": "0",
    "aux-param-depthclampmax": "65536",
    "aux-param-depthclampmin": "0",
    "aux-param-disparityshift": "0",
    "controls-autoexposure-auto": "True",
    "controls-autoexposure-manual": "8500",
    "controls-color-autoexposure-auto": "True",
    "controls-color-autoexposure-manual": "156",
    "controls-color-backlight-compensation": "0",
    "controls-color-brightness": "0",
    "controls-color-contrast": "50",
    "controls-color-gain": "64",
    "controls-color-gamma": "300",
    "controls-color-hue": "0",
    "controls-color-power-line-frequency": "3",
    "controls-color-saturation": "64",
    "controls-color-sharpness": "50",
    "controls-color-white-balance-auto": "True",
    "controls-color-white-balance-manual": "4600",
    "controls-depth-gain": "16",
    "controls-laserpower": "150",
    "controls-laserstate": "on",
    "ignoreSAD": "0",
    "param-autoexposure-setpoint": "1536",
    "param-censusenablereg-udiameter": "9",
    "param-censusenablereg-vdiameter": "9",
    "param-censususize": "9",
    "param-censusvsize": "9",
    "param-depthclampmax": "65536",
    "param-depthclampmin": "0",
    "param-depthunits": "1000",
    "param-disableraucolor": "0",
    "param-disablesadcolor": "0",
    "param-disablesadnormalize": "0",
    "param-disablesloleftcolor": "0",
    "param-disableslorightcolor": "0",
    "param-disparitymode": "0",
    "param-disparityshift": "0",
    "param-lambdaad": "800",
    "param-lambdacensus": "26",
    "param-leftrightthreshold": "24",
    "param-maxscorethreshb": "2047",
    "param-medianthreshold": "500",
    "param-minscorethresha": "1",
    "param-neighborthresh": "7",
    "param-raumine": "1",
    "param-rauminn": "1",
    "param-rauminnssum": "3",
    "param-raumins": "1",
    "param-rauminw": "1",
    "param-rauminwesum": "3",
    "param-regioncolorthresholdb": "0.0499022",
    "param-regioncolorthresholdg": "0.0499022",
    "param-regioncolorthresholdr": "0.0499022",
    "param-regionshrinku": "3",
    "param-regionshrinkv": "1",
    "param-robbinsmonrodecrement": "10",
    "param-robbinsmonroincrement": "10",
    "param-rsmdiffthreshold": "4",
    "param-rsmrauslodiffthreshold": "1",
    "param-rsmremovethreshold": "0.375",
    "param-scanlineedgetaub": "72",
    "param-scanlineedgetaug": "72",
    "param-scanlineedgetaur": "72",
    "param-scanlinep1": "60",
    "param-scanlinep1onediscon": "105",
    "param-scanlinep1twodiscon": "70",
    "param-scanlinep2": "342",
    "param-scanlinep2onediscon": "190",
    "param-scanlinep2twodiscon": "130",
    "param-secondpeakdelta": "325",
    "param-texturecountthresh": "0",
    "param-texturedifferencethresh": "0",
    "param-usersm": "1",
    "param-zunits": "1000",
    "stream-depth-format": "Z16",
    "stream-fps": "30",
    "stream-height": "720",
    "stream-width": "1280"
}
````

dorodnic, 2019-02-19:
Please take a look at #3275, there are some code snippets there
If I'm not mistaken the JSON you have attached is just the default configuration of the camera.  What parts of camera configuration you are trying to change by loading the JSON?

lvitters, 2019-02-19:
Sorry, I must have mistaken the JSON files, I edited my post to the correct one.

I'm mainly looking to enable post processing filters because the camera feed is not suitable for my needs without them. I don't think setting them in the Viewer and then getting the JSON saves those settings?

dorodnic, 2019-02-19:
Sorry, post-processing is not saved via this type of config file. The device knows nothing about the post-processing at the application layer. You should explicitly invoke post-processing filters in JNI layer.

lvitters, 2019-02-19:
Alright, thanks for the info!

closed 2019-02-19 by lvitters

spk921, 2019-05-08:
@AntoineWefit  What is "dev" on your code? What is the name space and how I could declare?

AntoineWefit, 2019-05-13:
@spk921 It is described [here](https://intelrealsense.github.io/librealsense/doxygen/classrs2_1_1device.html) and [here](https://github.com/IntelRealSense/librealsense/blob/79dda7f00ebde9564f1a2053b8ab95b7ba41ac69/include/librealsense2/hpp/rs_device.hpp).
You can declare it like so : 
`rs2::device dev;`

Hope this will help !