realsenseai/librealsense #3278: rs-pcl-color can't run

TouchDeeper, 2019-02-18:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400  | 
| Firmware Version                   |05.09.13.00| 
| Operating System & Version |    Ubuntu 16 | 
| Kernel Version (Linux Only)    |  4.15.0                                       | 
| Platform                                 | PC  |
| SDK Version                            | 2.8.1                         | 
| Language                            |  pcl                          | 
| Segment			|  Robot                        | 

### Issue Description
Hi,
I make the sdk2 by 
`cmake .. DCMAKE_BUILD_TYPE=Release  -DBUILD_EXAMPLES=true  -DBUILD_PCL_EXAMPLES=true` 
but the rs-pcl-color didn't make with the sdk2,all the bin generated is below,I can't find the rs-pcl-color.
[image]
Then  i make the rs-pcl-color separately，it make properly. But when i run,the error occur：

    /home/wang/CLionProjects/intelRealsense/pcl-color/cmake-build-debug/rs-pcl-color
    terminate called after throwing an instance of 'rs2::error'
      what():  Couldn't resolve requests

     Process finished with exit code 134 (interrupted by signal 6: SIGABRT)

I debug the code, when the code runs to `rs2::device selected_device = selection.get_device();`, the error occur.
Can you give me some advice?Thanks a lot!

dorodnic, 2019-02-18:
Could it be that the device is detected as USB2? You can check in the Viewer. 

rs-pcl-color was contributed to be built outside the main SDK build. There is an open pull-request trying to address it if you want to give it a try.

TouchDeeper, 2019-02-18:
oh,thanks you!
the viewer show the usb type is 2.1. the usb i plug in should be 3, thecolor of the interface is blue.Strangely,Is it a fake usb3?what a pity. But i can run other example properly with the same usb. Can the code be modified to run in usb2?

dorodnic, 2019-02-18:
The connection type really depends on many things - first, if you could share `dmesg` log after plugging the camera, it would help rule software issue. Next, often the cable is to blame. If you have another USB-C cable I'd give it a try. 
Stream configuration is defined here:
[rs-pcl-color.cpp:168-170](https://github.com/IntelRealSense/librealsense/blob/master/wrappers/pcl/pcl-color/rs-pcl-color.cpp#L168-L170).
You can use the Viewer to find resolutions that work in USB2. 
Newer firmware versions of D415 and D435 can offer color and IR streams via USB2, but USB3 is strongly recommended.

TouchDeeper, 2019-02-18:
Thank you for replying so quickly.
I think it is because i use  a USB-C that is not original. Can't it use any usb3-c?

dorodnic, 2019-02-18:
The camera will work with **most** cables and USB controllers. We do not require any specific manufacturer for accessories, but at the same time we can't test them all. There are discussions on the web about types of hardware that seem to work well for people.

TouchDeeper, 2019-02-19:
Thanks a lot!
I use the usb-c that is original,and the rs-pcl-color can run.

closed 2019-02-19 by TouchDeeper