realsenseai/librealsense #3281: D435i IMU Framerate

JBBee, 2019-02-18:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 05.11.01.00 | 
| Operating System & Version |   Win 10  | 
| Kernel Version (Linux Only)    |                                           | 
| Platform                                 |                                            |
| SDK Version                            |  2.18.1                                | 
| Language                            |  C++                          | 
| Segment			|  AR                         | 

### Issue Description
I'm seeing that the accelerometer frequency doesn't respect the frame rate given to enable_stream.  For example I have : 
`
motionConfig.enable_stream(RS2_STREAM_ACCEL, RS2_FORMAT_MOTION_XYZ32F, 250);
motionConfig.enable_stream(RS2_STREAM_GYRO, RS2_FORMAT_MOTION_XYZ32F, 400);
motionPipe->start(motionConfig, imuCallback);`

But the IMU accelerometer frequency is still ~63Hz.  Also, when I run the viewer and select 250 for the accelerometer framerate the stream info still shows it as ~63Hz.  

Is there any way to get the accelerometer to 250Hz?

Thanks.

RealSenseCustomerSupport, 2019-02-22:
----------------------------------------------

Hello @JBBee,

Have you tried calibrating the IMU on D435i camera?

D435i Calibration White Paper:
https://www.intel.com/content/dam/support/us/en/documents/emerging-technologies/intel-realsense-technology/RealSense_Depth_D435i_IMU_Calib.pdf

The FW you are using is up to date so that should not be an issue.

Also please verify that you are connected to a USB 3.0 port, the Viewer should display "3.1" or "3.2" in top left.  Or perhaps try a different USB port than the one you are using currently.

We can verify with Windows 10 / D435i / FW 5.11.1 / RealSense Viewer 2.18.1 the frame rate issue you report is not being seen.  See attached screenshot for verification.

Please update this ticket with your results after you have calibrated D435i IMU with white paper.
Thank you!

Attachment(s):
D415i_Win10_FPS.png - https://realsensesupport.intel.com/attachments/token/D2rIRdpHHOacJ1aEYUne9Z4Ht/?name=D415i_Win10_FPS.png

JBBee, 2019-03-01:
Hi @RealSenseCustomerSupport 

I have calibrated the IMU on the D435i previously but I re-calibrated it just now following the instructions in the white paper.  However, I'm still seeing strange framerates for the IMU streams as shown in the screenshot below.  The Viewer displays 3.2 in the top left too.

[image]

Thanks.

RealSenseCustomerSupport, 2019-03-15:
----------------------------------------------

Hi @JBBee,

Please try the latest librealsense and firmware and see how it works. If the issue still there, please enable librealsense log before realsense-viewer has IMU stream.

JBBee, 2019-03-28:
Hi @RealSenseCustomerSupport I tried again with fw version 5.11.01.100 and SDK version 2.19.2 but I'm still seeing the accelerometer FPS at 500Hz in the realsense viewer.  Below is the logfile if that's helpful.

This issue is no longer holding me up as we're now using the T265, and we're very happy with it.  So feel free to close this if you're unable to reproduce the issue.

[librealsense.log](https://github.com/IntelRealSense/librealsense/files/3019813/librealsense.log)

RealSenseCustomerSupport, 2019-03-29:
----------------------------------------------

From the log, I only can see 200Hz for gyro and 63Hz for accel (instead of 500Hz as you mentioned). Will further check if I can reproduce.

RealSenseCustomerSupport, 2019-03-29:
----------------------------------------------

This issue based on realsense-viewer 2.19.2 is still not observed.

closed 2019-04-01 by JBBee