realsenseai/librealsense #3282: Segmentation Fault when Running distance_to_object.ipynb

ShadowHunter0510, 2019-02-19:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Operating System & Version | Linux (Ubuntu 16) | 
| Platform                                 | Raspberry Pi | 
| Language                            | Python | 

### Issue Description
When running sample Python code such as [distance_to_object.ipynb](https://github.com/IntelRealSense/librealsense/blob/jupyter/notebooks/distance_to_object.ipynb), I get an error that simply says “Segmentation Fault” about half of the time that I run the program. Oftentimes it fixes itself when I rerun the program, but other times it requires for me to manually unplug the camera and replug it again to get rid of the problem. It either happens right before it says “Frames Captured” in the code, or if I am running the code in a loop, it would happen at the end of the current cycle in the loop.

[image]

RealSenseCustomerSupport, 2019-02-21:
----------------------------------------------

@ShadowHunter0510 What's the FW and librealsense version you're testing?

ShadowHunter0510, 2019-02-21:
Hi @RealSenseCustomerSupport 

The librealsense version that I’m using is 2.17.0, and the firmware version is 05.10.13.00.

RealSenseCustomerSupport, 2019-02-22:
----------------------------------------------

@ShadowHunter0510 Did you see this problem on PCs? Or you just got this issue on Raspberry  Pi?

ShadowHunter0510, 2019-02-22:
Hi @RealSenseCustomerSupport 

I’ve only been using a Raspberry Pi, so I’ve only been encountering these issues on that motherboard.

RealSenseCustomerSupport, 2019-02-25:
----------------------------------------------

@ShadowHunter0510 Could you please have a try on PC first? I tired on my PC and didn't get the issue. I'll try to set up a Raspberry Pi to see if any issue there.

ShadowHunter0510, 2019-02-25:
I think it might be something to do with the Raspberry Pi. In Issue #3303, another user confirmed that it could be causing all the problems? Can you tell me the results that you get?

RealSenseCustomerSupport, 2019-03-01:
----------------------------------------------

@ShadowHunter0510  Might be related to memory leak with python as mentioned in https://github.com/IntelRealSense/librealsense/issues/3341

RealSenseCustomerSupport, 2019-03-04:
----------------------------------------------

@ShadowHunter0510 I saw you have another ticket https://github.com/IntelRealSense/librealsense/issues/3180 is also reporting "Segmentation Fault". Seems duplicate? And I'm sorry I didn't find Raspberry Pi for the set up yet. You can keep an eye on https://github.com/IntelRealSense/librealsense/issues/3180 and https://github.com/IntelRealSense/librealsense/issues/3341 at first. Thanks!

RealSenseCustomerSupport, 2019-03-21:
----------------------------------------------

@ShadowHunter0510 As this is duplicate with https://github.com/IntelRealSense/librealsense/issues/3180 Let's track there and close this accordingly. Thanks!

closed 2019-03-28 by RealSenseCustomerSupport