realsenseai/librealsense #3283: Keeping Colors Constant in Depth Module

ShadowHunter0510, 2019-02-19:
### Issue Description
Is there any way to adjust the camera settings for the D435 so that the depth values are consistent with their corresponding colors? Currently, based on the distances of different objects in the frame, an rgb value of [0, 5, 255] could indicate a distance of 0.25 m or 0.50 m for example as the color bar changes. What I would like to do is for the colors to always convey the same distances instead of relative distances. Is this possible?

dorodnic, 2019-02-19:
Do you mean using the `colorizer`?
You can:
```cpp
rs2::colorizer colorize;
colorize.set_option(RS2_OPTION_COLOR_SCHEME, 2);
colorize.set_option(RS2_OPTION_HISTOGRAM_EQUALIZATION_ENABLED, 0);
colorize.set_option(RS2_OPTION_MIN_DISTANCE, 0);
colorize.set_option(RS2_OPTION_MAX_DISTANCE, 2.55); // max distance in meters
```
With this, when you get `[x, x, x]` you should be able to interpret it as x cm.
This is, however, very not accurate method to work with depth data.

ShadowHunter0510, 2019-02-19:
Hi @dorodnic 

Yes, that appears to be what I’m looking for. How can I convert that code into Python or where can I input that?

dorodnic, 2019-02-19:
The python wrapper should be one to one similar to C++, so something like:
```
rs2.colorizer colorize;
colorize.set_option(rs2.option.color_scheme, 2);
...
```

ShadowHunter0510, 2019-02-19:
Hi @dorodnic 

I’m sorry for the multitude of questions, but on the RealSense Viewer, I’m able to set the color scheme for the stereo module from “jet” to “white to black”. How do I make this setting default through code? Thank you for answering my previous question.

dorodnic, 2019-02-19:
What do you mean by default?
The viewer saves these settings, but in code you must configure it every time
Or you could change librealsense source code and make it default for your flavor of librealsense.dll

ShadowHunter0510, 2019-02-19:
Hi @dorodnic 

Never mind, that was a bad question. I’m not sure how much you know about NumPy, but how would you access the xy coordinates of a certain pixel? For example, how would I output the coordinates in (x,y) for the pixel with RGB value [a, a, a]?

lramati, 2019-02-21:
By using the get_data() function of a frame, you can get a standard NumPy
array of all the pixels, so the regular NumPy lookup functions will work.
There should be examples online of how to pull the index of a specific
value, but if I'm not mistaken (although I admit my NumPy is a bit rusty)
array.where() is the function you're looking for

dorodnic, 2019-02-25:
Hope this helped :)

closed 2019-02-25 by dorodnic