realsenseai/librealsense #3284: How to connect Intel's Realsense D435 camera to an Intel's ARRIA10 SoC development kit?

makilambu, 2019-02-19:
We have tried connecting Intel Realsense 435i camera to an Ubuntu desktop and it works as expected but we would like to know is there any support or README regarding how to connect the camera to a ARM running on Intel's ARRIA10 SoC Development board with Ubuntu 16.04 armhf base OS and USB 2.0 ( we are ok with the low resolution of the camera)?

MartyG-RealSense, 2019-02-19:
If you are using an SoC instead of an FPGA,, I'm guessing you are using an Arria 10 SX?

According to a specification chart of the SX range, they use a USB OTG (On The Go) mini-USB port.  These are designed to allow full-size USB devices to be connected to a small device via an adapter.  

You should be able to connect your D435i to an OTG port with an adapter.  The 400 Series can run on single-board computing devices - Intel's own Click online store sells bundle deals of a camera and 'Up Board' SoC.   

There have been a small number of reports with problems regarding stability of power supply to the 400 Series camera when using it with an OTG port on single-board devices such as Up Board and Intel Compute Stick.  There's only a handful of such reports though and you may not have any trouble at all.  I just mention it so you are aware of the possibility.

If you encounter problems with installing Librealsense on the armhf OS, Librealsense has a couple of methods that enable it to be used with any flavor of Linux.

https://github.com/IntelRealSense/librealsense/issues/1132#issuecomment-363572666

dorodnic, 2019-02-19:
We have also added more documentation since #1132 -
[libuvc_installation.md](https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md)

closed 2019-03-07 by RealSenseCustomerSupport