realsenseai/librealsense #3287: Is there a clone method for the rs2::frame?

dongxuanlb, 2019-02-19:
My scene is one thread use pipeline capture rs::frame, and more then one threads consume the rs::frame; consider thread synchronization issues，all consume thread in the sync block need deep copy the producer rs::frame first.

in opencv, I can use cv::Mat.clone(), is there a clone method for the rs2::frame?

dorodnic, 2019-02-20:
Hi @dongxuanlb 
Why do you need to deep copy the frame object? 
You can just pass `rs2::frame`s around between different threads via thread safe `rs2::frame_queue` primitive. You might want to call `frame.keep()` if processing takes too long, but other then that there should be no need for clone.
[do-processing-on-a-background-thread](https://github.com/IntelRealSense/librealsense/wiki/API-How-To#do-processing-on-a-background-thread)
Alternatively, you can convert frames to `cv::Mat` and proceed from there.

dongxuanlb, 2019-02-23:
Hi @dorodnic
deep clone is because I have more than one threads to process a frame concurrently.

I'm not export for the c++, I just refer to the mjpeg-streamer, and now I convert the frame to Mat first, and use clone method, it works well!

dorodnic, 2019-02-24:
Hi @dongxuanlb 
If you have multiple threads all reading from the same frame, there is no threading issue with that.
You should only be concerned if at any point you forcefully remove `(const)` qualifier from frame data.

dorodnic, 2019-03-10:
Hopefully resolved
Feel free to open follow-up issues if needed

closed 2019-03-10 by dorodnic

blakat360, 2020-01-29:
I'm trying to store a depth frame to compare with another frame at a later time. Is a .clone() method not the best option?