realsenseai/librealsense #3288: Advanced Mode : Android API

ssakhavi, 2019-02-19:
@Matkatz @Dorondic

In the current code for unrooted devices, there is no support for advanced mode. 

Based on the RealSense  C API, in order to set the advanced mode settings, there is a need to receive a device handle. 

Receiving the device handle takes around 3 seconds and in addition to other preparations for the streaming, which leads to a large delay from start to streaming. 

A sample of our code can be seen here:
Advanced mode JNI:
https://github.com/KroniKare/RealSense-NewAndroid/blob/KroniKare-Update/src/android/jni/advancedmode.c

Java Activity:
https://github.com/KroniKare/RealSense-NewAndroid/blob/KroniKare-Update/wrappers/android/examples/capture/src/main/java/com/intel/realsense/capture/MainActivityWithImageView.java

ssakhavi, 2019-03-01:
@dorodnic @matkatz 

The problem has been solved after updating the code to the current development branch.

dorodnic, 2019-04-29:
We have also added the ability to load presets in JSON format to JNI wrapper in the last release. This should be sufficient to modify all advanced mode settings.

closed 2019-04-29 by dorodnic