realsenseai/librealsense #3289: Align color to depth without holes in the resulting image

EmmanuelCos, 2019-02-19:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Operating System & Version |   Win10 |
| Language                            |  C++                     | 

Hello,
I am trying to extract the depth and color separatly. To do so, I first aligne color to depth (and not the opposite), as I want to have a 1280 * 720 resolution at the end. 
I've manage to do that, but the RGB image I get have black holes when there are holes in the depth image (I attach one example)

Is there a way to have the same image, but without these holes?

[image]

Thanks and regards

dorodnic, 2019-02-19:
Most likely black pixels correspond to pixels in depth camera viewport that are occluded from RGB sensor by objects in the scene.

EmmanuelCos, 2019-02-20:
Yes I know but my will is to have an aligned RGB pictures without holes... Do we know what transformation is made on the original RGB? Is there a resize or something that I could apply to the original one, and in this way I wouldn't have these holes and the RGB would be aligned to the depth?

dorodnic, 2019-02-20:
[This is the transformation](https://github.com/IntelRealSense/librealsense/blob/master/src/proc/align.cpp#L19) but I don't see how it can help. You are asking to get RGB values for pixels that the RGB sensors **did not see**.

darrell-rg, 2019-02-21:
If you use the realsense ros driver and rviz you can fill in the occluded pixels with data from previous frames.   You would have to move the mouse out of the way and then put it back while holding the camera still.   I don't know if it works with a moving camera I have not tried. 

 https://github.com/intel-ros/realsense

closed 2019-03-18 by RealSenseCustomerSupport