realsenseai/librealsense #3292: Build instructions for newer kernels

landersson, 2019-02-20:
For various reasons, I'd like to use the RealSense SDK on a laptop running Ubuntu 18.10 and Linux kernel 4.18. If possible, I would also like to run it on an ARM based embedded system (NanoPI M4).

The libraries and tools seem to compile fine on 18.10, but I'm unsure about how to apply and build the required kernel patches. Would it be possible to provide basic instructions for how to apply and build kernel patches for newer kernels? 

Is it likely that minor modifications to the ```scripts/patch-ubuntu-kernel-4.16.sh``` script will be able to successfully build custom kernel modules for kernel 4.16? If so, what would be the suggested modifications?

dorodnic, 2019-02-20:
We certainly have plans to continue to expand the list of supported kernels as we did during last two years. At the moment we are looking into problems with 4.19 kernel.
Also, we are improving the [libuvc alternative](https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md) for platforms where patching is impossible (currently working on adding IMU support to enable D435i)

landersson, 2019-02-20:
Ok, that sounds good. Any chance you could provide a very brief overview of the steps I should attempt in order to try making it work on the 4.18 kernel?

dorodnic, 2019-02-20:
Couple of questions - 
1. Do you need to use the D435i?
2. Do you need access to per-frame metadata (actual exposure, gain, etc...)

landersson, 2019-02-21:
1. Yes, the device we have in mind is a D435i.

2.  Since this will be used for computer vision purposes, per frame exposure etc would be useful, but it's not critical.

dorodnic, 2019-02-21:
The overall flow is always:
1. Download source code for the Kernel you are using.
2. Navigate to the relevant folder and apply one or more of the patches.
3. If patches cannot apply cleanly, make minor modifications to make them apply successfully
4.a. Assuming your kernel has modules, run `make` under relevant module directory, this will produce new `.ko` binary file. This file can be loaded using `modprobe`.
4.b. If your kernel is "monolithic" and does not support dynamically loaded kernel modules, you will have to build the entire kernel and boot into it (there are guides on how to do that)

> As always, don't forget to apply udev-rules, otherwise you will get `Permissions Denied`

Under [`/scripts/`](https://github.com/IntelRealSense/librealsense/tree/master/scripts) we have many versions of patches modified to apply cleanly on different distributions.
You don't necessarily need all of the patches. 
If you are based on 4.18, you will already have depth, IR and RGB streaming and hardware timestamps. To make frame metadata work, you need to apply [realsense-metadata-ubuntu-xenial-v4.16.patch](https://github.com/IntelRealSense/librealsense/blob/master/scripts/realsense-metadata-ubuntu-xenial-v4.16.patch).
For D435i motion module, you are recommended to apply [realsense-hid-ubuntu-xenial-v4.16.patch](https://github.com/IntelRealSense/librealsense/blob/master/scripts/realsense-hid-ubuntu-xenial-v4.16.patch). I think it should work without, but again, you will not get proper timestamps for sure without a patch.

landersson, 2019-02-21:
Thanks @dorodnic, that's very helpful. I will try it as soon as I get a chance.

fubsan, 2019-03-12:
> We certainly have plans to continue to expand the list of supported kernels as we did during last two years. At the moment we are looking into problems with 4.19 kernel.

Hi @dorodnic ,

How are the above mentioned plans going. Is there an expected roadmap/release in sight?
We're also kind of stuck with 4.14 on arch (which switched to 4.19 lts wise). We would like to take 4.19 lts into use but have heavy stability issues with it in combo with d435.

landersson, 2019-03-27:
I just successfully patched and cross-compiled a modified 4.4.154 kernel for the NanoPI M4 SOC using the patch files in /scripts/, so feel free to close this issue.

closed 2019-04-11 by RealSenseCustomerSupport

jkeys-ecg-nmsu, 2019-06-05:
@dorodnic are you still looking at kernel 4.19? I'm unable to install the realsense packages on Chrome OS (Google Pixel Slate) and trying to find a solution.

lantianhaohh, 2022-11-01:
Hi @dorodnic , My question at #11050 ,Based on my problem description, I would like to ask how I should go about kernel patching, I want to explore kernel patching, which is helpful for my project.Thank you.