realsenseai/librealsense #3293: A question about IMU

AdamPengG, 2019-02-20:
Hi author,
I want to know where is the imu position on the D4 board and the distance from the imu to RGB imager sensor or two gray imager sensors. The three cameras' coordinate system.
Can you tell me the IMU's  noise and random walk bias and it's right-handed coordinate system or left-handed coordinate system.
The RT (rotation and translation ) about the imu with three cameras.

dorodnic, 2019-02-20:
https://github.com/IntelRealSense/librealsense/blob/master/doc/imu_and_tracking_sensors.md

AdamPengG, 2019-02-20:
I only find  imu's coordinate in this link.Can you tell me more?

Shashika007, 2019-02-21:
@AdamPengG You can find the extrinsic calibration of IMU and Camera using Kalibr calibration. IMU's noise and random walk bias you can find it using imu_utils (https://github.com/gaowenliang/imu_utils). Also, you should check the extrinsic topics which published from realsense. You can get some idea of the rotation and translation of three cameras with respect to depth frame which is also aligned with IMU frame. (https://github.com/IntelRealSense/librealsense/blob/master/doc/imu_and_tracking_sensors.md)

closed 2019-04-09 by RealSenseCustomerSupport