realsenseai/librealsense #3294: Hello

Carlbouakl, 2019-02-20:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

MartyG-RealSense, 2019-02-20:
How the 400 Series cameras work is that for most stream modes, the camera data is rectified in the camera, in its Vision Processor D4 hardware component.  The rectified data is then passed through the USB cable to the computer.  

The exception to this is the 'Y16' IR mode, which is unrectified.  This is because it is designed to be used for camera calibration rather than everyday general use in applications.  For this reason, there are only a couple of resolutions that support Y16, and a limited range of FPS speeds.

1920x1080, 25 FPS, with depth stream disabled
1920x1080, 15 FPS, with depth diabled
960x540, 25 FPS, with depth disabled
960x540, 15 FPS, with depth disabled

You can also only use Y16 mode with IR on its own, and not have a depth stream active at the same time.

Here is an example of such a setup in the RealSense Viewer program.  In the Viewer, the red streaming activation button is ghosted out in grey unless settings are configured that meet the strict criteria listed above.

[image]

The link below leads to a user-contributed Python script for displaying both IR streams.  Substitute your own desired settings for the stream into the values in the brackets in the stream config instructions.  For example, for left and right IR streams with 960x540 resolution and 25 FPS, I believe the changes should be something like this:

config.enable_stream(rs.stream.infrared, 1, 960, 540, rs.format.y16, 25)
config.enable_stream(rs.stream.infrared, 2, 960, 540, rs.format.y16, 25)

Carlbouakl, 2019-02-21:
Thank you for your reply
Your solution includes getting the infrared images, whereas I need the raw ones.

MartyG-RealSense, 2019-02-21:
Given that Y16 is not rectified, it is probably close to what you would get from a true rawstream without calibration.  

I recall that the camera has a few streams mode called RAW, such as 'RAW10'.  

https://github.com/IntelRealSense/librealsense/blob/5e73f7bb906a3cbec8ae43e888f182cc56c18692/include/librealsense2/h/rs_sensor.h?language=en_US#L65

Others have asked before about whether RAW10 could be used to get a raw stream, and the Intel support staff's reply at the Intel Support website was that although the mode existed, it was not exposed to the end-user.

Carlbouakl, 2019-02-21:
I specifically need the raw images (not necessarily rectified) for my application.

I'm guessing it wont work with a D435 camera?

MartyG-RealSense, 2019-02-21:
When a mode is described as not being 'exposed', in general it typically means that although it is supported in the firmware driver, access to it is not supported in the SDK software.  So access to it may be able to be added by the SDK developers, but it is not available for use until then.  So it is likely a problem related to the SDK software and not a specific 400 Series camera model.

pedrombmachado, 2019-02-21:
Please check below a script for getting the 2 NIR images:
```
import pyrealsense2 as rs
import numpy as np
import cv2

points = rs.points()
pipeline = rs.pipeline()
config = rs.config()
config.enable_stream(rs.stream.infrared, 1, 640, 480, rs.format.y8, 30)
config.enable_stream(rs.stream.infrared, 2, 640, 480, rs.format.y8, 30)
profile = pipeline.start(config)

try:
    while True:
        frames = pipeline.wait_for_frames()
        nir_lf_frame = frames.get_infrared_frame(1)
        nir_rg_frame = frames.get_infrared_frame(2)
        if not nir_lf_frame or not nir_rg_frame:
            continue
        nir_lf_image = np.asanyarray(nir_lf_frame.get_data())
        nir_rg_image = np.asanyarray(nir_rg_frame.get_data())
        # horizontal stack
        image=np.hstack((nir_lf_image,nir_rg_image))
        cv2.namedWindow('NIR images (left, right)', cv2.WINDOW_AUTOSIZE)
        cv2.imshow('IR Example', image)
        key = cv2.waitKey(1)
        # Press esc or 'q' to close the image window
        if key & 0xFF == ord('q') or key == 27:
            cv2.destroyAllWindows()
            break
finally:
    pipeline.stop()
```

RealSenseCustomerSupport, 2019-02-25:
----------------------------------------------

Hi Carlbouakl,

Given what MartyG-RealSense and pedrombmachado have explained & suggested, is there anything else needed?

Thanks!

Carlbouakl, 2019-02-26:
Getting stereo raw images won't work because its an RGBD camera and not a stereo camera.
I modified my code and used the infrared images.

Thanks everyone!

closed 2019-02-26 by Carlbouakl