realsenseai/librealsense #3296: Hardware Notification: USB CAM overflow

HiOtto, 2019-02-20:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i  | 
| Firmware Version                   | 05.10.13.00 | 
| Operating System & Version  |  Linux (Ubuntu 16.04)  | 
| Kernel Version (Linux Only)    |  4.4.38+  | 
| Platform                                  | TX2 JETPACK3.2.1 L4T28.2.1 |
| SDK Version                           | 2.18.1 | 

### Issue Description
CAN GET:
depth image, imu data
CAN NOT GET:
2 infrared images, RGB image

I tested the steam with realsense_viewer and rs_camera.launch(in ros), result are attached

MartyG-RealSense, 2019-02-20:
Do you still get an overflow if you reduce the resolution to a lower setting, please?

HiOtto, 2019-02-21:
> Do you still get an overflow if you reduce the resolution to a lower setting, please?

I reduced both resolution and frame rate, then neither get infrared images nor RGB image
[image]

HiOtto, 2019-02-21:
> Do you still get an overflow if you reduce the resolution to a lower setting, please?

HiOtto, 2019-02-22:
xxx@XXX:~$ lsmod
Module                  Size  Used by
hid_sensor_accel_3d     5405  0
hid_sensor_gyro_3d      5392  0
hid_sensor_trigger      3597  4 hid_sensor_gyro_3d,hid_sensor_accel_3d
hid_sensor_iio_common     5931  3 hid_sensor_trigger,hid_sensor_gyro_3d,hid_sensor_accel_3d
industrialio_triggered_buffer     1912  2 hid_sensor_gyro_3d,hid_sensor_accel_3d
kfifo_buf               3467  1 industrialio_triggered_buffer
hid_sensor_hub         12069  4 hid_sensor_trigger,hid_sensor_gyro_3d,hid_sensor_accel_3d,hid_sensor_iio_common
uvcvideo               78097  0
videobuf2_vmalloc       5624  1 uvcvideo
ipt_MASQUERADE          2115  1
nf_nat_masquerade_ipv4     2931  1 ipt_MASQUERADE
iptable_nat             2285  1
nf_nat_ipv4             6554  1 iptable_nat
xt_addrtype             3298  2
iptable_filter          2119  1
ip_tables              18322  2 iptable_filter,iptable_nat
xt_conntrack            3551  1
nf_nat                 16285  2 nf_nat_ipv4,nf_nat_masquerade_ipv4
br_netfilter           13923  0
overlay                33899  0
pci_tegra              60337  0

xxx@XXX:~$ rs-enumerate-devices
 Device info: 
    Name                          : 	Intel RealSense D435I
    Serial Number                 : 	843112072054
    Firmware Version              : 	05.10.13.00
255.255.255.255
    Recommended Firmware Version  : 	05.10.03.00
    Physical Port                 : 	/sys/devices/3530000.xhci/usb2/2-2/2-2:1.0/video4linux/video0
    Debug Op Code                 : 	15
    Advanced Mode                 : 	YES
    Product Id                    : 	0B3A
    Usb Type Descriptor           : 	3.2

Stream Profiles supported by Stereo Module
 Supported modes:                                      stream     resolution fps   format   
    Infrared 1	  1280x800	@ 30Hz	Y8
    Infrared 2	  1280x800	@ 30Hz	Y8
    Infrared 1	  1280x800	@ 25Hz	Y16
    Infrared 2	  1280x800	@ 25Hz	Y16
    Infrared 1	  1280x800	@ 15Hz	Y16
    Infrared 2	  1280x800	@ 15Hz	Y16
    Infrared 2	  1280x800	@ 15Hz	Y8
    Infrared 1	  1280x800	@ 15Hz	Y8
    Infrared 1	  1280x720	@ 30Hz	Y8
    Infrared 2	  1280x720	@ 30Hz	Y8
    Infrared 1	  1280x720	@ 15Hz	Y8
    Infrared 2	  1280x720	@ 15Hz	Y8
    Infrared 1	  1280x720	@ 6Hz	Y8
    Infrared 2	  1280x720	@ 6Hz	Y8
    Infrared 1	  848x480	@ 90Hz	Y8
    Infrared 2	  848x480	@ 90Hz	Y8
    Infrared 1	  848x480	@ 60Hz	Y8
    Infrared 2	  848x480	@ 60Hz	Y8
    Infrared 2	  848x480	@ 30Hz	Y8
    Infrared 1	  848x480	@ 30Hz	Y8
    Infrared 2	  848x480	@ 15Hz	Y8
    Infrared 1	  848x480	@ 15Hz	Y8
    Infrared 1	  848x480	@ 6Hz	Y8
    Infrared 2	  848x480	@ 6Hz	Y8
    Infrared 1	  640x480	@ 90Hz	Y8
    Infrared 2	  640x480	@ 90Hz	Y8
    Infrared 2	  640x480	@ 60Hz	Y8
    Infrared 1	  640x480	@ 60Hz	Y8
    Infrared 2	  640x480	@ 30Hz	Y8
    Infrared 1	  640x480	@ 30Hz	Y8
    Infrared 2	  640x480	@ 15Hz	Y8
[... 243 lines cut ...]
    Color	  320x180	@ 60Hz	RGB8
    Color	  320x180	@ 60Hz	Y16
    Color	  320x180	@ 60Hz	BGRA8
    Color	  320x180	@ 60Hz	RGBA8
    Color	  320x180	@ 60Hz	BGR8
    Color	  320x180	@ 60Hz	YUYV
    Color	  320x180	@ 30Hz	RGB8
    Color	  320x180	@ 30Hz	Y16
    Color	  320x180	@ 30Hz	BGRA8
    Color	  320x180	@ 30Hz	RGBA8
    Color	  320x180	@ 30Hz	BGR8
    Color	  320x180	@ 30Hz	YUYV
    Color	  320x180	@ 6Hz	RGB8
    Color	  320x180	@ 6Hz	Y16
    Color	  320x180	@ 6Hz	BGRA8
    Color	  320x180	@ 6Hz	RGBA8
    Color	  320x180	@ 6Hz	BGR8
    Color	  320x180	@ 6Hz	YUYV

Stream Profiles supported by Motion Module
 Supported modes:                                      stream     resolution fps   format   
    Gyro	@ 200Hz	MOTION_XYZ32F
    Gyro	@ 400Hz	MOTION_XYZ32F
    Accel	@ 63Hz	MOTION_XYZ32F
    Accel	@ 250Hz	MOTION_XYZ32F

HiOtto, 2019-02-22:
Lastest Debug Info:
https://github.com/IntelRealSense/librealsense/issues/3312#issue-413461807

Close this earlier one

closed 2019-02-22 by HiOtto

jungladicitta, 2019-08-08:
@HiOtto 
how did you solve this issue?
i have similar

mvoelk, 2022-02-21:
I had similar issues, after hardware reset and firmware update via realsense-viewer, the errors are gone.

HappySamuel, 2022-03-09:
Hi @mvoelk 
Which firmware version are you using? I am also facing the same issue, wish to see if the issues will disappear after applying the firmware version you used.

Best,
Samuel

mvoelk, 2022-03-10:
@HappySamuel most recent, I assume, 05.13.00.50 in my case, but I'm still having issues with ROS and have to do hardware resets from time to time.

HappySamuel, 2022-03-11:
Hi @mvoelk 
Thanks for the version info. Besides, how to perform hardware resets?

Best,
Samuel

MartyG-RealSense, 2022-03-11:
Hi @HappySamuel  The hardware reset option in the RealSense Viewer can be found by going to the **More** option near the top of the Viewer's options side-panel and selecting **Hardware Reset** from its drop-down menu.