realsenseai/librealsense #3297: The tool rs-convert does not terminate

frankfalse, 2019-02-20:
I use the tool rs-convert to get png file from bag file.
But any time I start the program it does't terminate. I'm using a Linux System so to terminate the program I must to press the combination of keys Ctrl-C.
Someone have the same problem?

* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

frankfalse, 2019-02-21:
I add a screenshot which shows the execution end of the tool when it extract frames from the file bag 20190220_145430_880.bag.
The screenshot shows that the bag file contains 45 frame:

[image]

To close the application I need to press Ctrl+C.

Thank you
Frank

frankfalse, 2019-02-22:
May be that I have found a work-around for my problem.
I note that the tool rs-convert doesn't terminate because the condition for exit from the main while(true) is:
if (frameset[0].get_frame_number() < frameNumber) {
  break;
}
This condition is verified **when the file bag finishes and the program start to get the first frame of the bag file**.
In that moment the program exits from the main loop but it doesn't terminate.
The work-around consists to continue playing the file bag and, after the frame number 1, read the frame number 2 and the frame number 3 too, but without write anything on the disk (this means that I don't call the function stbi_write_png())
With this _strange_ modified the tool stops its execution!

Someone can explain this behavior?

Thanks

YangJiao1996, 2019-02-24:
Hi Frank, that is interesting. I believe that it is also the weird `if` statement you mention that causes the problem in both your issue and #3208. 
I noticed that @dorodnic referred this issue in a recent push request, is this issue solved in #3318?

dorodnic, 2019-02-25:
@YangJiao1996, yes, should be solved and part of a release in a day or two.
The solution is more of a work around, but it seems to get the job done, and I want to unblock this before diving deeper into the root cause of the issue.

frankfalse, 2019-02-27:
Thank you very much now all it works.

RealSenseCustomerSupport, 2019-02-27:
----------------------------------------------

Hi frankfalse,

Glad to know that fix from dorodnic works for you.

Thanks!

closed 2019-05-29 by RealSenseCustomerSupport