realsenseai/librealsense #3298: Unreal Engine 4.21 compilation error: deletion of pointer to incomplete type, no destructor called

Simon-Thompson, 2019-02-20:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 05.09.13.00 | 
| Operating System & Version |   Windows 10 v. 1803  | 
| Platform                                 | PC |
| SDK Version                            |  2.16.3.312 | 
| Language                            |  C++ | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
When trying to update the realsense wrapper for unreal engine to work with 4.21, the following error is outputted in VS Community 2017: 
"Error C4150 deletion of pointer to incomplete type 'rs2::points'; no destructor called" 
Same error repeats for types 'rs2::pointcloud;, 'rs2::align', 'rs2::device', 'rs2::pipeline' and 'FDynamicTexture'.

I'm having trouble fixing this, as it seems to be that UniquePtr.h in the engine's source code is throwing the error because the above types aren't defined. When trying to compile the same plugin with 4.20, I get no errors, and the code seems the same. I'm nearly at a working version of the plugin with 4.21, could anyone help?

I've made changes to the RealSense.Build.cs file to fix compilation errors with 4.21, see the attached zip for that file. 

To reproduce this issue, do the following steps:
 1. Create a blank c++ project with Unreal Engine 4.21
 2. Follow the instructions on https://github.com/IntelRealSense/librealsense/tree/master/wrappers/unrealengine4 to install the RealSense plugin
3. Delete the RuntimeMeshComponent plugin folder included in the previous step's download, and replace it with this: https://github.com/Moddingear/RuntimeMeshComponent. (It's a more up-to-date version of the same plugin that supports ue 4.21)
4. Modify the RealSense.Build.cs file with the changes in the attached file, or simply replace with the attached file.
5. Build the project

[RealSense.Build.cs.zip](https://github.com/IntelRealSense/librealsense/files/2886234/RealSense.Build.cs.zip)

Kabutak, 2019-05-10:
Running into a similar issue trying to make the wrapper compile for Unreal Engine 4.22 and SDK 2.21

I think I've eliminated most of the problems, but I'm still working on this one and a few other type-declaration issues. Will update if I find a good solution, but I may just scrap this attempt if it drags on for too long.

Simon-Thompson, 2019-05-10:
Thanks for the update Kabutak! I've had to put it on the backburner since I posted this, so any help is appreciated ^^

Kabutak, 2019-05-10:
I finally managed to resolve all of the issues after plugging away at it all morning. Most of the migration was simply macro redefinition, but a number of headers had to be fixed up or relocated as well. Most notably `RealSenseInspector.h` needed a few headers added into it. I ended up dumping a reference to the PCH in there to make it easier. [This link](https://docs.unrealengine.com/en-us/Programming/BuildTools/UnrealBuildTool/IWYU) may help provide more information. If I get time, I may try cleaning it up and opening a pull request so that others can at least use it as a reference.

Kabutak, 2019-05-10:
@Simon-Thompson I have cleaned up my changes and made a pull request (#3956). Hopefully that helps you if you come back to this stuff later on.

Simon-Thompson, 2019-05-10:
Beautiful! Thanks

closed 2019-05-10 by Simon-Thompson