realsenseai/librealsense #3299: D415 color camera: all options have value '6'

Nomadluap, 2019-02-20:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 05.09.02.00 | 
| Operating System & Version |  Ubuntu 16.04  | 
| Kernel Version (Linux Only)    |          4.4.0-142-generic                      | 
| Platform                                 | PC  |
| SDK Version                            |  2.18.0                          | 
| Language                            | c++                        | 

### Issue Description
I have  a Realsense D415 camera connected to my computer. When I try to set the exposure or other settings in the realsense-viewer program, all of the options seem to be stuck at the value '6' for the RGB camera. 

I see the following output from rs-enumerate-devices: 

```
$ tools/enumerate-devices/rs-enumerate-devices -o
 Device info: 
    Name                          :     Intel RealSense D415
    Serial Number                 :     822512061463
    Firmware Version              :     05.09.02.00
    Recommended Firmware Version  :     05.10.03.00
    Physical Port                 : 
    Debug Op Code                 :     15
    Advanced Mode                 :     YES
    Product Id                    :     0AD3

Options for Stereo Module
 Supported options:                                    min        max       step  default  
    Exposure                                           : 20   ... 166000      20    33000     
    Gain                                               : 6    ... 6           6     6         
    Enable Auto Exposure                               : 0    ... 1           1     1         
    Enable Auto White Balance                          : 0    ... 1           1     0         
    Visual Preset                                      : 0    ... 6           1     0         
    Laser Power                                        : 0    ... 360         30    150       
    Emitter Enabled                                    : 0    ... 2           1     1         
    Frames Queue Size                                  : 0    ... 32          1     16        
    Error Polling Enabled                              : 0    ... 1           1     0         
    Output Trigger Enabled                             : 0    ... 1           1     0         
    Depth Units                                        : 0.0001... 0.01        1e-06 0.001     
    Stereo Baseline                                    : 55.0903... 55.0903     0     55.0903   

Options for RGB Camera
 Supported options:                                    min        max       step  default  
    Backlight Compensation                             : 6    ... 6           6     6         
    Brightness                                         : 6    ... 6           6     6         
    Contrast                                           : 6    ... 6           6     6         
    Exposure                                           : 6    ... 6           6     6         
    Gain                                               : 6    ... 6           6     6         
    Gamma                                              : 6    ... 6           6     6         
    Hue                                                : 6    ... 6           6     6         
    Saturation                                         : 6    ... 6           6     6         
    Sharpness                                          : 6    ... 6           6     6         
    White Balance                                      : 6    ... 6           6     6         
    Enable Auto Exposure                               : 0    ... 1           1     1         
    Enable Auto White Balance                          : 6    ... 6           6     6         
    Frames Queue Size                                  : 0    ... 32          1     16        
    Power Line Frequency                               : 6    ... 6           6     6         
    Auto Exposure Priority                             : 6    ... 6           6     6         

[Snip]
```

Is there a reason why all the parameter values for the RGB camera are stuck at 6?

Nomadluap, 2019-02-20:
Updating the camera firmware to version 5.11.01 does not seem to fix the issue.

dorodnic, 2019-02-21:
Any warnings / errors in the dmesg log? This is certainly abnormal

Nomadluap, 2019-02-21:
Here is the output from my dmesg after plugging in the camera (with the USB cable supplied in the box) and running `rs-enumerate-devices`: 

```

[ 5225.138587] usb 2-2: new SuperSpeed USB device number 16 using xhci_hcd
[ 5225.154812] usb 2-2: New USB device found, idVendor=8086, idProduct=0ad3
[ 5225.154821] usb 2-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 5225.154827] usb 2-2: Product: Intel(R) RealSense(TM) Depth Camera 415 
[ 5225.154832] usb 2-2: Manufacturer: Intel(R) RealSense(TM) Depth Camera 415 
[ 5225.154837] usb 2-2: SerialNumber: 828313020766
[ 5225.156076] uvcvideo: Unknown video format 00000050-0000-0010-8000-00aa00389b71
[ 5225.156090] uvcvideo: Unknown video format 00000032-0000-0010-8000-00aa00389b71
[ 5225.156221] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 415  (8086:0ad3)
[ 5225.158267] uvcvideo: Unable to create debugfs 2-16 directory.
[ 5225.158616] input: Intel(R) RealSense(TM) Depth Ca as /devices/pci0000:00/0000:00:14.0/usb2/2-2/2-2:1.0/input/input1217
[ 5225.159513] uvcvideo: Unknown video format 36315752-1a66-a242-9065-d01814a8ef8a
[ 5225.159529] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 415  (8086:0ad3)
[ 5225.160601] uvcvideo: Unable to create debugfs 2-16 directory.
[ 5254.213298] uvcvideo: Unknown video format 00000050-0000-0010-8000-00aa00389b71
[ 5254.213315] uvcvideo: Unknown video format 00000032-0000-0010-8000-00aa00389b71
[ 5254.213439] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 415  (8086:0ad3)
[ 5254.215551] uvcvideo: Unable to create debugfs 2-16 directory.
[ 5254.215641] input: Intel(R) RealSense(TM) Depth Ca as /devices/pci0000:00/0000:00:14.0/usb2/2-2/2-2:1.0/input/input1218
[ 5254.273328] uvcvideo: Unknown video format 36315752-1a66-a242-9065-d01814a8ef8a
[ 5254.273333] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 415  (8086:0ad3)
[ 5254.274196] uvcvideo: Unable to create debugfs 2-16 directory.
```

RealSenseCustomerSupport, 2019-02-27:
----------------------------------------------

@Nomadluap  How about the realsenseviewer? Can you adjust these color parameters on realsenseviewer?

RealSenseCustomerSupport, 2019-03-01:
----------------------------------------------

@Nomadluap Any update? Looking forward to your update. Thanks!

Nomadluap, 2019-03-01:
Sorry for the delay! In Realsense viewer, the controls were also stuck at 6 for almost all values. Attempting to change the value using either the slider or text edit control would have no effect.

RealSenseCustomerSupport, 2019-03-04:
----------------------------------------------

@Nomadluap Does this D415 work normally before? Or this is the 1st time you tried it and saw the problem?
Could you please also try FW 5.10.6 to see if still same issue?

RealSenseCustomerSupport, 2019-03-07:
----------------------------------------------

@Nomadluap Any update?

Nomadluap, 2019-03-07:
I've found that after updating my operating system to Ubuntu 18.04 and re-building the realsense-viewer program, the camera works without issue (Using firmware 5.11.01). 

Running the cameras on other Ubuntu 16.04 systems also had no issues. 

My computer had a strange softaware setup, so it's not out of the question for the problem to be localized entirely to my computer. 

I'll mark this issue as closed for now since I can't seem to reproduce on other computers or operating systems.

closed 2019-03-07 by Nomadluap

barabas6river, 2019-05-01:
I am currently seeing this issue on a mac with a d435:

Device info: 
    Name                          : 	Intel RealSense D435
    Serial Number                 : 	801212070604
    Firmware Version              : 	05.10.06.00
    Recommended Firmware Version  : 	05.10.03.00
    Physical Port                 : 	
    Debug Op Code                 : 	15
    Advanced Mode                 : 	YES
    Product Id                    : 	0B07
    Usb Type Descriptor           : 	3.2

Options for Stereo Module
 Supported options:                min        max       step  default  
    Exposure                       : 1    ... 165000      1     8500      
    Gain                           : 6    ... 6           6     6         
    Enable Auto Exposure           : 0    ... 1           1     1         
    Visual Preset                  : 0    ... 6           1     0         
    Laser Power                    : 0    ... 360         30    150       
    Emitter Enabled                : 0    ... 1           1     1         
    Frames Queue Size              : 0    ... 32          1     16        
    Error Polling Enabled          : 0    ... 1           1     0         
    Output Trigger Enabled         : 0    ... 1           1     0         
    Depth Units                    : 0.0001... 0.01        1e-06 0.001     
    Stereo Baseline                : 50.0007... 50.0007     0     50.0007   
    Inter Cam Sync Mode            : 0    ... 2           0     1         

Options for RGB Camera
 Supported options:                min        max       step  default  
    Backlight Compensation         : 6    ... 6           6     6         
    Brightness                     : 6    ... 6           6     6         
    Contrast                       : 6    ... 6           6     6         
    Exposure                       : 6    ... 6           6     6         
    Gain                           : 6    ... 6           6     6         
    Gamma                          : 6    ... 6           6     6         
    Hue                            : 6    ... 6           6     6         
    Saturation                     : 6    ... 6           6     6         
    Sharpness                      : 6    ... 6           6     6         
    White Balance                  : 6    ... 6           6     6         
    Enable Auto Exposure           : 0    ... 1           1     1         
    Enable Auto White Balance      : 6    ... 6           6     6         
    Frames Queue Size              : 0    ... 32          1     16        
    Power Line Frequency           : 6    ... 6           6     6         
    Auto Exposure Priority         : 6    ... 6           6     6         

@RealSenseCustomerSupport Is there anything I can dump to give you more info?