realsenseai/librealsense #3301: Issue with the Realsense viewer on Ubuntu 16.04

makilambu, 2019-02-21:
Required Info 	
Camera Model                            -   D435i
Firmware Version 	                   -    5.11.01
Operating System & Version      -    Ubuntu 16.04
Kernel Version (Linux Only) 	   -    4.15.0-45-generic
Platform                                      -    PC
USB version                                -    USB 3.2
SDK Version 	                           -    2.18.0

After following the installation setup given in this page "https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md", I rebooted the server and tried to start with realsense viewer I get an error message stating that frame did not arrive. Here are the screen shot of the same:

[image]
[image]
[image]
[image]
[image]
when I do a dmesg and look for uvcvideo, I get these errors. 
[image]

But when i plug the camera into USB2.1 hub, i get stereo image alone without any error but it doesn't show the RGB or Motion knobs.

dorodnic, 2019-02-21:
Hi @makilambu 
Please try the 2.18.1 software. No need to re-patch. 
This was a bug in 2.18, where we assumed 5.11.1 firmware would support something it did not yet.

makilambu, 2019-02-22:
I tried to checkout 2.18.1 from the git and compiled and installed it. Now the issue is with realsense viewer too, the camera is not getting detected. Attached a picture of realsense viewer image. 

[image]

To check whether everything is sane with the system, I tried to check whether libusb has installed properly along with modprobe of uvcvideo to check whether realsense tag comes: 

[image]

[image]

Then I tried to check the USB port is working or not by using lsinput and the output seems to show the event of USB activity and it seems to be working 

[image]

Further went on with sudo udevadm monitor --udev to check the activity and here it shows it adds and removes the camera. 

[image]

Is there anything that I am missing to build it from the source?

ev-mp, 2019-02-25:
@makilambu , 
Are you sure you've build and tried v2.**18**.1 and not 2.8.1 as appear in the screenshot ?

If not - please share the following:
- Kernel log during camera connect event (`dmesg -wT`).
- `lsusb` output when the camera is connected
- `rs-enumerate-devices` output

RealSenseCustomerSupport, 2019-02-26:
----------------------------------------------

Hi makilambu,

I tried build RS 2.18.1 from source, and tested with D435i (FW5.11.1). There is no issue.
As ev-mp pointed out, can you please check your RS source code and then provide the asked logs if still with issue.

Thanks!

makilambu, 2019-03-01:
I reinstalled the SDK and it is working now as expected. Thanks for the support. I would like to close this thread

closed 2019-03-01 by makilambu