realsenseai/librealsense #3302: imx435i: accl, gyroscope: FIXED: negative value are shown as unsigned on ****_***_raw sysfs.

arskama, 2019-02-21:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | imx435i | 
| Operating System & Version |  linux ubuntu 16 or 18  | 
| Kernel Version (Linux Only)    |  kernel < 4.20                                       | 
| Platform                                 | PC  |

### Issue Description
accelerometer and gyroscope
sysfs in_acc_x_raw and others, are reporting unsigned int for signed int.
so the basic operation in_acc_x_raw/ in_acc_scale is not working.

Fix exists in: https://lkml.org/lkml/2018/12/14/598, released in kernel 4.20, so if you want to get the correct values on Linux, upgrade your kernel.

rakuco, 2019-02-21:
To clarify: librealsense itself works fine since it parses `scan_elements/in_accel_*_type` to read the information, but anyone reading the sysfs values directly might need a heads-up in the documentation.

RealSenseCustomerSupport, 2019-03-04:
----------------------------------------------

Hi @arskama, @rakuco,

Can you elaborate more if any improvement required for librealsense?

arskama, 2019-03-04:
Hi @RealSenseCustomerSupport 

I don't think so.
As @rakuco mentioned, librealsense should be fine.

closed 2019-03-12 by RealSenseCustomerSupport