realsenseai/librealsense #3303: Raspberry Pi Settings

ShadowHunter0510, 2019-02-21:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | 05.10.13.00 | 
| Operating System & Version | Ubuntu 16 | 
| Platform                                 | Raspberry Pi |

### Issue Description

What are the recommended settings for a Raspberry Pi? Since the computer only supports USB 2.0, the camera has limited functionality. How can I squeeze out the maximum quality from the Raspberry Pi?

ban1080, 2019-02-21:
Run, then run away more, I'm using Pi 3B+ and running these cameras on the USB2.0 interface is the most unstable thing ever, tested both the D415 and D435. 

D415 has many more resolution/framerate combos than the D435. Currently, I'm running 1280x720 @ 6FPS and using the D435 (this option is also available on the D415), the 848×480 option (which maximises accuracy for D435) is not available on USB2.0.

I'm only requesting one frame every few seconds. Keep in mind you cannot control the IR sensor (nor its power) on the USB 2.0 interface.

The camera works 1/3 the time, the other 2/3 is comprised of random segmentation faults, resource unavailable errors and others. 

To get the max performance I'd recommend maxing out the amount of available RAM through raspi-config (which means no desktop). 

This is an rs-enumerate dump for the D435:
[image]

RealSenseCustomerSupport, 2019-03-04:
----------------------------------------------

Hi @ShadowHunter0510,

Still need any support for this topic to run D400 camera on Pi?

RealSenseCustomerSupport, 2019-04-02:
----------------------------------------------

I will close this

closed 2019-04-09 by RealSenseCustomerSupport