realsenseai/librealsense #3304: rs-capture example link error from glfw with Visual Studio 2017 on Windows10

cookiedog, 2019-02-21:
Required Info | |
------------ | -------------
Camera Model	| D435i
Firmware Version	| 05.11.01.00
Operating System & Version	| Win 10 64bits
Platform	|   PC
SDK Version	| 2.18.1
Language	| Visual Studio 2017(15.9.7) / C++ 

### Issue Description

I copy the sample code from rs-capture.cpp to main.cpp, but the project complete was fail.
I have checked the example.hpp and glfw3.h is loading, but the error is no improvement.
Is there any resolution for this issue or readme or any similar problems others faced that can help with get going?

- Code
```
// License: Apache 2.0. See LICENSE file in root directory.
// Copyright(c) 2017 Intel Corporation. All Rights Reserved.

#include <librealsense2/rs.hpp> // Include RealSense Cross Platform API
#include "example.hpp"          // Include short list of convenience functions for rendering

// Capture Example demonstrates how to
// capture depth and color video streams and render them to the screen
int main(int argc, char * argv[]) try
{
    rs2::log_to_console(RS2_LOG_SEVERITY_ERROR);
    // Create a simple OpenGL window for rendering:
    window app(1280, 720, "RealSense Capture Example");

    // Declare depth colorizer for pretty visualization of depth data
    rs2::colorizer color_map;
    // Declare rates printer for showing streaming rates of the enabled streams.
    rs2::rates_printer printer;

    // Declare RealSense pipeline, encapsulating the actual device and sensors
    rs2::pipeline pipe;

    // Start streaming with default recommended configuration
    // The default video configuration contains Depth and Color streams
    // If a device is capable to stream IMU data, both Gyro and Accelerometer are enabled by default
    pipe.start();

    while (app) // Application still alive?
    {
        rs2::frameset data = pipe.wait_for_frames().    // Wait for next set of frames from the camera
                             apply_filter(printer).     // Print each enabled stream frame rate
                             apply_filter(color_map);   // Find and colorize the depth data

        // The show method, when applied on frameset, break it to frames and upload each frame into a gl textures
        // Each texture is displayed on different viewport according to it's stream unique id
        app.show(data);
    }

    return EXIT_SUCCESS;
}
catch (const rs2::error & e)
{
    std::cerr << "RealSense error calling " << e.get_failed_function() << "(" << e.get_failed_args() << "):\n    " << e.what() << std::endl;
    return EXIT_FAILURE;
}
catch (const std::exception& e)
{
    std::cerr << e.what() << std::endl;
    return EXIT_FAILURE;
}
```

- Output

```

1>------ Has started construction: Project: hello-world, Configuration: Release x64 ------
1>main.cpp
1>main.obj : error LNK2001: Unresolved external symbol glfwSwapBuffers
1>main.obj : error LNK2001: Unresolved external symbol __imp_glPointSize
1>main.obj : error LNK2001: Unresolved external symbol __imp_glPopMatrix
1>main.obj : error LNK2001: Unresolved external symbol __imp_glViewport
1>main.obj : error LNK2001: Unresolved external symbol __imp_glEnableClientState
1>main.obj : error LNK2001: Unresolved external symbol __imp_glClear
1>main.obj : error LNK2001: Unresolved external symbol glfwGetWindowUserPointer
1>main.obj : error LNK2001: Unresolved external symbol __imp_glPopAttrib
1>main.obj : error LNK2001: Unresolved external symbol __imp_glRotated
1>main.obj : error LNK2001: Unresolved external symbol __imp_glBindTexture
1>main.obj : error LNK2001: Unresolved external symbol __imp_glGenTextures
1>main.obj : error LNK2001: Unresolved external symbol __imp_glVertexPointer
1>main.obj : error LNK2001: Unresolved external symbol __imp_glEnable
1>main.obj : error LNK2001: Unresolved external symbol __imp_glVertex3f
1>main.obj : error LNK2001: Unresolved external symbol __imp_glEnd
1>main.obj : error LNK2001: Unresolved external symbol glfwGetFramebufferSize
1>main.obj : error LNK2001: Unresolved external symbol __imp_glTexParameterfv
1>main.obj : error LNK2001: Unresolved external symbol glfwTerminate
1>main.obj : error LNK2001: Unresolved external symbol gluLookAt
1>main.obj : error LNK2001: Unresolved external symbol __imp_glTexCoord2f
1>main.obj : error LNK2001: Unresolved external symbol __imp_glDisableClientState
1>main.obj : error LNK2001: Unresolved external symbol glfwGetWindowSize
1>main.obj : error LNK2001: Unresolved external symbol glfwSetMouseButtonCallback
1>main.obj : error LNK2001: Unresolved external symbol __imp_glMatrixMode
1>main.obj : error LNK2001: Unresolved external symbol __imp_glTexCoord2fv
1>main.obj : error LNK2001: Unresolved external symbol glfwPollEvents
1>main.obj : error LNK2001: Unresolved external symbol glfwInit
1>main.obj : error LNK2001: Unresolved external symbol __imp_glLineWidth
1>main.obj : error LNK2001: Unresolved external symbol glfwSetKeyCallback
1>main.obj : error LNK2001: Unresolved external symbol __imp_glColor3f
1>main.obj : error LNK2001: Unresolved external symbol glfwSetScrollCallback
1>main.obj : error LNK2001: Unresolved external symbol glfwCreateWindow
1>main.obj : error LNK2001: Unresolved external symbol __imp_glLoadIdentity
1>main.obj : error LNK2001: Unresolved external symbol __imp_glVertex3fv
1>main.obj : error LNK2001: Unresolved external symbol glfwMakeContextCurrent
1>main.obj : error LNK2001: Unresolved external symbol __imp_glTexParameteri
1>main.obj : error LNK2001: Unresolved external symbol glfwSetWindowUserPointer
1>main.obj : error LNK2001: Unresolved external symbol __imp_glBegin
1>main.obj : error LNK2001: Unresolved external symbol glfwDestroyWindow
1>main.obj : error LNK2001: Unresolved external symbol __imp_glClearColor
1>main.obj : error LNK2001: Unresolved external symbol __imp_glDrawArrays
1>main.obj : error LNK2001: Unresolved external symbol __imp_glTranslatef
1>main.obj : error LNK2001: Unresolved external symbol __imp_glTexImage2D
1>main.obj : error LNK2001: Unresolved external symbol __imp_glVertex2f
1>main.obj : error LNK2001: Unresolved external symbol __imp_glRotatef
1>main.obj : error LNK2001: Unresolved external symbol glfwSetCursorPosCallback
1>main.obj : error LNK2001: Unresolved external symbol __imp_glVertex2d
1>main.obj : error LNK2001: Unresolved external symbol gluPerspective
1>main.obj : error LNK2001: Unresolved external symbol __imp_glDisable
1>main.obj : error LNK2001: Unresolved external symbol __imp_glPushMatrix
1>main.obj : error LNK2001: Unresolved external symbol __imp_glGetFloatv
1>main.obj : error LNK2001: Unresolved external symbol __imp_glPixelStorei
1>main.obj : error LNK2001: Unresolved external symbol glfwWindowShouldClose
1>main.obj : error LNK2001: Unresolved external symbol __imp_glOrtho
1>main.obj : error LNK2001: Unresolved external symbol __imp_glGetError
1>main.obj : error LNK2001: Unresolved external symbol __imp_glPushAttrib
1>C:\Program Files (x86)\Intel RealSense SDK 2.0\samples\x64\Release\hello-world.exe : fatal error LNK1120: 56 unresolved external symbols
1>Project "hello-world.vcxproj" Construction completed -- failed.
========== Built: 0 Success, 1 Failure, 0 Latest, 0 Skip ==========  
```

darrell-rg, 2019-02-21:
The linker can't find glfw3.lib.  To fix, go to project properties and point the linker to the right file (path may be different on your machine):
[image]
I got glfw3.lib  as part of the CMake librealsense BUILD_GRAPHICAL_EXAMPLES ,  but you can also get it somewhere else.  
[image]

cookiedog, 2019-02-25:
@darrell-rg Thanks for your help! I solved the problem according to your method.

closed 2019-03-05 by RealSenseCustomerSupport