realsenseai/librealsense #3305: Is it possible to retrieve ply  coordinates in meters?

Woodstock94, 2019-02-21:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / **D400** } | 
| Operating System & Version |   {Win (**8.1**/10) / Linux (Ubuntu 14/16/17) / MacOS  |                     | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/**python**/unity }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>
I'm able to export scenes in ply files using pyrealsense 2, I wonder if it's possible for instance to calculate the distance between 2 points belonging to the pointcloud and retrieve the information in meters [m].

Cheers

RealSenseCustomerSupport, 2019-03-05:
----------------------------------------------

Hi Woodstock94,

You may want to check out measurement example, if not yet.
https://github.com/IntelRealSense/librealsense/tree/master/examples/measure
It would be helpful.

Thanks!

RealSenseCustomerSupport, 2019-03-21:
----------------------------------------------

Hi Woodstock94,

Do you still need help on this one?
If no, we are going to close it.

Thanks!

Woodstock94, 2019-03-21:
sorry, i forgot to reply.
Thanks for the help.

closed 2019-03-28 by RealSenseCustomerSupport