realsenseai/librealsense #3306: D435 Image stabilization

tRosenflanz, 2019-02-21:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435 } | 
| Firmware Version                   | '05.11.01.00'| 
| Operating System & Version |   Linux (Ubuntu 14/16/17)  | 
| Kernel Version (Linux Only)    |  4.15.0-45-generic                                         | 
| Platform                                 | PC|
| SDK Version                            | 2.16                          | 
| Language                            |  python                           | 
| Segment			|  others                         | 

Hello, 

This is just a generic question about image stabilization capabilities of D435 (or D435i) cameras. In our use case there is a lot of device vibration which can causes  serious distortions in ordinary cameras:

1) Does this camera have any internal image stabilization hardware and what are some of the specs you can provide about it? 
2) Does the SDK have any image stabilization algorithms which we can use? 
3) Do you happen to have any recommendations for external stabilizers? 

Thanks,
Tim

MartyG-RealSense, 2019-02-21:
In a 2018 Intel online webinar session about RealSense, a member of the audience asked the presenter the question of how often the cameras have to be calibrated to keep point clouds aligned during vibration.  They were told that in general you should not have to, although you may perhaps need to do so in harsher conditions.  The stereo algorithm is designed to be robust, they said.

If the camera is going to be experiencing a lot of vibration, you could also consider using USB cables with screw-locks so the cable can be firmly attached to the pair of screw-holes either side of the camera's USB port.  this should help prevent the cable from working its way out of the USB port during high vibration or movement of other parts near the camera.  this subject is discussed in the link below.

https://forums.intel.com/s/question/0D50P0000490S80SAE/where-can-i-buy-a-usb-typec-cable-with-screws-on-d435?language=en_US

Vibration may become more of an issue if you plan to use SLAM.  On Intel's new RealSense T265 Tracking Camera with IMU, 'wheel odometry' is used as a stabilizing mechanism.

RealSenseCustomerSupport, 2019-02-26:
----------------------------------------------

Hi

Thanks for the good question. Unfortunately current D400 camera doesn't have any OIS system or software IS feature.

Personally i think D43x might have the potential in the future since it is global shutter imager. The IS algorithm should be less difficult than rolling shutter imager like D41x. In your case, a very high frame rate camera would be helpful. i.e. 120FPS or higher should be much better.

Thanks

tRosenflanz, 2019-03-06:
Thank you for the responses. We purchased a few D435 specifically because of the global shutter so hopefully it works well in our use case.

closed 2019-03-06 by tRosenflanz