realsenseai/librealsense #3308: USB SCP Overflow D415. and data resolution falls, decrease in fps.

ankith001, 2019-02-21:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

ankith001, 2019-02-21:
I have an overflow of frames. Resolution of depth data falls by half from 1280x720 to 640x360 and fps is fluctuating from 21 to 30 fps.

darrell-rg, 2019-02-21:
I had this problem on Nvidia Jetson Xavier.  The solution that worked for me was to use an external PCIe to USB3 card.

ankith001, 2019-02-21:
@darrell-rg  I am using USB3 card. But still, I have the same issue.

MartyG-RealSense, 2019-02-21:
Could you check please if the overflow errors disappear if you use a lower resolution, please?

ankith001, 2019-02-21:
[image]
[image]

pncanas333, 2019-02-22:
I have experiment this fluctuation in the fps on Windows, in Ubuntu it disappears. 
Other thing you could try is turning off the Auto Exposure Priority, this can be found on the RGB options under the controls section. Maybe it is also due to the auto exposure setting....

RealSenseCustomerSupport, 2019-02-27:
----------------------------------------------

@ankith001 What's your testing platform and FW version of the camera?

RealSenseCustomerSupport, 2019-03-01:
----------------------------------------------

@ankith001 Could you please update the FW version you got this issue? Thanks!

ankith001, 2019-03-01:
I am using on Windows and FW version is 2.18.

RealSenseCustomerSupport, 2019-03-04:
----------------------------------------------

@ankith001 Thanks for the update! Please note that 2.18 is the librealsense version. Please check the camera FW(Firmware) version via rs-enumerate or realsenseviewer "info" tab. It should be something like 05.10.06.00.

RealSenseCustomerSupport, 2019-03-07:
----------------------------------------------

@ankith001 Any update? Do you get the FW version?

RealSenseCustomerSupport, 2019-03-11:
----------------------------------------------

@ankith001  Do you still have the problem with latest FW and librealsense? Looking forward to your update. Thanks!

RealSenseCustomerSupport, 2019-03-15:
----------------------------------------------

@ankith001 Did you still need help on this? Please update. Thanks!

pavloblindnology, 2019-03-19:
Have very severe "incomplete video frame detected!" problem on Xavier and D435 with latest library (2.19.1) and FW (05.11.01.100), either using recent Jetsonhacks patching or without. On laptop everything works fine. In 1280x720, 15fps, depth+left+right+color, the error message shows up constantly. 30fps is unbearable at all. Even 640x480 crashes .
After some time "USB SCP overflow" can appear too.

closed 2019-03-22 by RealSenseCustomerSupport