realsenseai/librealsense #3311: Implement Spatial and Temporal Filters in Matlab

TOMAxxHAWK, 2019-02-22:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415                                                    | 
| Firmware Version                   | 5.11.1| 
| Operating System & Version |   Win 10  | 
| Kernel Version (Linux Only)    |  -                                        | 
| Platform                                 | PC |
| SDK Version                            |  v2.17.1                    | 
| Language                            |  Matlab 2018b                        | 
| Segment			|  -                      | 

### Issue Description

I am trying to develop a depth measuring program in Matlab, to verify a distance over a few pixels as function of the time. Noise seems to be the biggest problem for me, because every frame has an entirely pattern. I was trying to implement the Temporal and Spatial filter into my program. I see that they have been developed but I am stuck in how to use them. I only have experience with matlab due to my physics study. Matlab is very useful to gather a lot of data and use them for furthur analyses. But replicating these filters in Matlab takes too much processing time.

Does anybody have a clue on how to use them and also where to place them? You can use the depth_example() as a way to show where I should put it. So I have a clear idea where to place it inside my own program.

lramati, 2019-02-22:
The Matlab API is very nearly one to one with the C++ API, so the C++
example on post-processing filters should be a good jumping off point for
understanding how to use them in Matlab

TOMAxxHAWK, 2019-02-22:
Thank you, I managed to implement it thanks to the examples. I was already close thanks to a friend who had who understood the language. Something I noticed is that the temporal filter and spatial filter both miss a 'end' statement. An 'end' should be placed above the line with: this = this@realsense.filter(out);. In both cases it is missing.

closed 2019-03-11 by RealSenseCustomerSupport