realsenseai/librealsense #3313: D435 Exposure Time Behaviour USB 2.0 vs USB 3.0

ban1080, 2019-02-22:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435  | 
| Firmware Version                   | 5.11.01 | 
| Operating System & Version |   Raspbian  | 
| Kernel Version (Linux Only)    |  4.16.*                                         | 
| Platform                                 | Raspberry Pi |
| SDK Version                            |          2.18.1                 | 
| Language                            |  C++                          | 
| Segment			|  others                          | 

### Issue Description
Setting exposure time to 1000us and taking photos on USB 2.0 and USB 3.0 yields different results. USB 3.0 shows the intended effect whereby motion blur is largely eliminated. 

On USB 2.0
[image]

On USB 3.0
[image]

ban1080, 2019-02-22:
When the exposure time is set to 173us on USB 2.0 this happens:
[image]

Setting the camera to Auto-Exposure while on USB 2.0 yields slightly better results but is very unusable for our purposes.

ban1080, 2019-02-23:
As suspected, the camera works fine when taking at 640x480@30 fps. The increase in fps seems to have done something...

[image]

We noticed a similar effect on the D410, whereby increasing the FPS (through accepting lower resolution) significantly reduced the effect of rolling shutter. We suspected that a lower fps somehow caused the camera to have greater delay between reading each row.

We had thought that getting the D430 would eliminate that problem due to the global shutter, but it introduces a different problem of motion blur...

RealSenseCustomerSupport, 2019-02-26:
----------------------------------------------

Hi

Blurry image is always visible if exposure time is too long, no matter the imager is global shutter or rolling shutter. One of the advantages of global shutter is that it is better than rolling shutter in terms of distortion on moving object. With that being said, you should see similar blur effect if imager was set at same exposure time.

Higher frame rate means the maximum exposure time is shorter than lower frame rate. For example, the maximum exposure time is 33ms under 30FPS, 66ms under 15FPS ....

The exposure time 173us as you mentioned for 640x480@30fps doesn't seem to be correct to me - it shouldn't be so bright. I guess you might have set it to 173ms, but that will make camera frame rate drop to <6fps.

Under manual exposure mode, if the image still looks dark, can you try increase the Gain (try 64, 128, 248 max)? (NOTE: more gain, more noise)

Thanks

ban1080, 2019-02-27:
Hi

Thank you for your reply.

The camera was at 1000us while tested during 640x480@30fps, not 173us.

The issue isn't related to exposure time nor gain, the issue is related to the camera behaving differently when used with a USB 2.0 connection relative to USB 3.0 under the same resolution, framerate and exposure settings.

USB 2.0 yields incorrect results (e.g. reducing exposure time **does not** reduce motion blur), while USB 3.0 yields correct results (e.g. reducing exposure time reduces motion blur). 

The really bright image has an exposure time of **173us**, the camera behaves like that when used with USB 2.0, but not with USB 3.0. Setting it with 174us on USB 2.0 yields semi-expected behaviour (image becomes very dark, but motion blur is still very strong). 

Please advise,
Thank you

RealSenseCustomerSupport, 2019-02-28:
----------------------------------------------

Hi

That's the trade-off between USB2 bandwidth vs imager performance.
Due to USB2 bandwidth limitation, the imager data readout time has to be extended 4x as compared under USB3. Since global shutter imager requires fast readout to ensure the shutter performance. Longer readout time causes lower shutter efficiency. That's why under USB2, you see the "ghost tail" effect.

Thanks

ban1080, 2019-02-28:
That makes sense, It'd be nice to formally document all the limitations that the USB 2.0 interface has, it would've saved time and money.

Thank you for your reply.

closed 2019-02-28 by ban1080