realsenseai/librealsense #3315: Announcement: Firmware version 5.11.1.100 for RealSense 400 Series cameras now available

MartyG-RealSense, 2019-02-23:
Hi everyone,

An incremental firmware update (from version 5.11.1 to 5.11.1.**100**) is now available from the link below.

https://downloadcenter.intel.com/download/28568/Latest-Firmware-for-Intel-RealSense-D400-Product-Family?v=t

Release notes for this firmware version are not yet available.

dorodnic, 2019-02-28:
Thanks @MartyG-RealSense 
I wanted to offer some clarification since current firmware page is a bit confusing:

[5.11.1.100](https://downloadcenter.intel.com/download/28568/Latest-Firmware-for-Intel-RealSense-D400-Product-Family?v=t) is a hot-fix to latest stable firmware (that being 5.11.1 right now). It is mostly identical in functionality to 5.11.1.

[5.11.4](https://downloadcenter.intel.com/download/28573/Latest-Firmware-for-Intel-RealSense-D400-Product-Family) is newer, but potentially less stable, firmware. One of the new capabilities it introduces is [emitter toggle on-off](https://github.com/IntelRealSense/librealsense/issues/1299) for the D435 and the 35i. In case of the 35i this + #3066 can allow better SLAM by taking advantage of full infrared field of view.

MartyG-RealSense, 2019-02-28:
Yes, I figured that was the situation with .100, since I couldn't see any new additions listed in the firmware release notes.  Thanks for the info on that and 11.4!  I'll add the info about the toggle to the copy of the announcement on Intel Support.

closed 2019-03-10 by dorodnic