realsenseai/librealsense #3316: roslaunch realsense2_camera rs_camera.launch , rostopics are listed properly but topics are empty

da-piyushpatel, 2019-02-23:
WARNING [140262693140224] (backend-hid.cpp:692) HID set_power 1 failed for /sys/devices/pci0000:00/0000:00:14.0/usb2/2-2/2-2:1.5/0003:8086:0B3A.000D/HID-SENSOR-200073.1.auto/iio:device0/buffer/enable
[ INFO] [1550942541.279994694]: gyro stream is enabled - fps: 400
 23/02 22:52:21,280 WARNING [140262904284928] (backend-hid.cpp:692) HID set_power 1 failed for /sys/devices/pci0000:00/0000:00:14.0/usb2/2-2/2-2:1.5/0003:8086:0B3A.000D/HID-SENSOR-200076.2.auto/iio:device1/buffer/enable
[ INFO] [1550942541.280522692]: accel stream is enabled - fps: 250
[ INFO] [1550942541.284063188]: num_filters: 0
[ INFO] [1550942541.285118251]: RealSense Node Is Up!
[ INFO] [1550942541.285229257]: Setting Dynamic reconfig parameters.
 23/02 22:52:21,290 WARNING [140263832737664] (types.cpp:57) hwmon command 0x7d failed. Error type: Invalid Command (-1).
[ERROR] [1550942541.290331265]: An exception has been thrown: hwmon command 0x7d failed. Error type: Invalid Command (-1).
[FATAL] [1550942541.412978724]: Failed to load nodelet '/camera/realsense2_camera` of type `realsense2_camera/RealSenseNodeFactory` to manager `realsense2_camera_manager'
[camera/realsense2_camera_manager-2] process has died [pid 7457, exit code -11, cmd /opt/ros/kinetic/lib/nodelet/nodelet manager __name:=realsense2_camera_manager __log:=/home/ppatel/.ros/log/930dddca-378f-11e9-9aa1-f8633f6c3c43/camera-realsense2_camera_manager-2.log].
log file: /home/ppatel/.ros/log/930dddca-378f-11e9-9aa1-f8633f6c3c43/camera-realsense2_camera_manager-2*.log
[camera/realsense2_camera-3] process has died [pid 7458, exit code 255, cmd /opt/ros/kinetic/lib/nodelet/nodelet load realsense2_camera/RealSenseNodeFactory realsense2_camera_manager __name:=realsense2_camera __log:=/home/ppatel/.ros/log/930dddca-378f-11e9-9aa1-f8633f6c3c43/camera-realsense2_camera-3.log].
log file: /home/ppatel/.ros/log/930dddca-378f-11e9-9aa1-f8633f6c3c43/camera-realsense2_camera-3*.log

dorodnic, 2019-02-24:
What versions of ROS wrapper, librealsense and firmware version are you using?

da-piyushpatel, 2019-02-25:
Thanks I figured out the problem it was due to mismatch of versions.

closed 2019-02-25 by da-piyushpatel