realsenseai/librealsense #3319: error occur when device.pipeline.poll_for_frame() using box dimensioner multicam demo

su11jing11, 2019-02-25:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D435  | 
| Firmware Version                   | 05.11.01.00 | 
| Operating System & Version |   {Win 10  | 
| Platform                                 | PC  |
| SDK Version                            | pyrealsense2                       | 
| Language                            |  python                        | 

### Issue Description
Hi I am trying to run the box dimensioner multicam demo, however get the following error:

Traceback (most recent call last):
  File "C:/Users/su11j/PycharmProjects/SerialAndSocket/venv/box_dimensioner_multicam/box_dimensioner_multicam_demo.py", line 143, in <module>
    run_demo()
  File "C:/Users/su11j/PycharmProjects/SerialAndSocket/venv/box_dimensioner_multicam/box_dimensioner_multicam_demo.py", line 52, in run_demo
    frames = device_manager.poll_frames()
  File "C:\Users\su11j\PycharmProjects\SerialAndSocket\venv\box_dimensioner_multicam\realsense_device_manager.py", line 212, in poll_frames
    device.pipeline.try_wait_for_frames(frameset)
TypeError: try_wait_for_frames(): incompatible function arguments. The following argument types are supported:
    1. (self: pyrealsense2.pyrealsense2.pipeline, timeout_ms: int=5000) -> Tuple[bool, pyrealsense2.pyrealsense2.composite_frame]

Invoked with: <pyrealsense2.pyrealsense2.pipeline object at 0x0B9DB820>, <pyrealsense2.pyrealsense2.composite_frame object at 0x0B9DB8E0>

I find that there is someone asked exactly the same question, and the advice is to use wait_for_frames(). I tried it, but still get imcompatible function argument error. Could you guys help for this? Thanks.

RealSenseCustomerSupport, 2019-03-04:
----------------------------------------------

Hi @su11jing11,

1. Your D435 firmware is 5.11.1.0, right?
2. Do you build the pyrealsense2 library from source? You can try to install the prebuilt library ($ pip install pyrealsense2), ex: 2.18.1.xxx.
3. Use "wait_for_frames" in realsense_device_manager.py

RealSenseCustomerSupport, 2019-03-05:
----------------------------------------------

Hi @su11jing11,

Please also refer to #3325 to see if the PR fix contributed by @darrell-rg can work for you.

closed 2019-04-19 by RealSenseCustomerSupport