realsenseai/librealsense #3320: Can pipe.start(callback) and pipe.start(cfg) are used at the same time?

TouchDeeper, 2019-02-25:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435  | 
| Firmware Version                   |05.11.01.00| 
| Operating System & Version |    Ubuntu 16 | 
| Kernel Version (Linux Only)    |  4.15.0                                       | 
| Platform                                 | PC  |
| SDK Version                            | 2.8.1                         | 
| Language                            |  C++                         | 
| Segment			|  Robot                        | 

### Issue Description
Hi,
I read the example of callback, and have some doubts about this. Can pipe.start(callback) and pipe.start(cfg) are used at the same time or Can config cfg set the callback function?
Can you give me some advice? Thanks a lot!

dorodnic, 2019-02-25:
Yes. `pipe.start(cfg, callback)` should work

TouchDeeper, 2019-02-25:
Yard！ I'm too stupid.
You are right. Thanks a lot!

closed 2019-02-25 by TouchDeeper