realsenseai/librealsense #3323: Announcement:  Intel Aero UAV drone platform is being discontinued

MartyG-RealSense, 2019-02-25:
Hi everyone,

For the information of customers using RealSense cameras with the Intel Aero UAV drone platform, I would like to make you aware that that Intel Aero platform and support for it is being discontinued in June 2019, and new purchases of parts for Aero from the Intel Click store is no longer possible.

https://forums.intel.com/s/question/0D50P00004CfP2bSAF/intel-aero-platform-for-uavs-discontinuation-notice?language=en_US

dorodnic, 2019-03-10:
Thanks @MartyG-RealSense 
This is not directly related to RealSense, but as you mentioned might have been useful to some of our customers.

closed 2019-03-10 by dorodnic

MartyG-RealSense, 2019-03-10:
RealSense has been officially paired with the Intel Ready To Fly drone for years, with a caseless version of the RealSense R200 camera sold in the UAV section of the Click store as part of the RealSense Vision Accessory Kit for Intel drones and Intel Aero Compute Board single-board computer.  So there is a direct connection between RealSense and Intel UAV products.  That was the rationale behind posting an announcement about it here.