realsenseai/librealsense #3325: Need help running box_dimensioner_multicam.py

MaryBD, 2019-02-25:
Hello, I am following the above code and I got this error:
Traceback (most recent call last):
  File "C:/Users/Maryam/PycharmProjects/multicam/box_dimensioner_multicam_demo.py", line 141, in <module>
    run_demo()
  File "C:/Users/Maryam/PycharmProjects/multicam/box_dimensioner_multicam_demo.py", line 52, in run_demo
    frames = device_manager.poll_frames()
  File "C:\Users\Maryam\PycharmProjects\multicam\realsense_device_manager.py", line 195, in poll_frames
    device.pipeline.poll_for_frames(frameset)
TypeError: poll_for_frames(): incompatible function arguments. The following argument types are supported:
    1. (self: pyrealsense2.pyrealsense2.pipeline) -> pyrealsense2.pyrealsense2.composite_frame

Invoked with: <pyrealsense2.pyrealsense2.pipeline object at 0x000001837FE0BC38>, <pyrealsense2.pyrealsense2.composite_frame object at 0x000001837FE0BDC0>

Any advice?
thanks.

darrell-rg, 2019-02-25:
I made a PR https://github.com/IntelRealSense/librealsense/pull/2949 that fixes this, it has not been accepted yet.

Please try it and let me know if it works for you.

MaryBD, 2019-02-26:
Thanks. There is an error using that, what should be the input of function? it returns this error when I tried to use it as a function in the main code:
    frames = poll_frames()
TypeError: poll_frames() missing 1 required positional argument: 'self'

and this error when I use it as a class:

   for (serial, device) in self._enabled_devices.items():
AttributeError: 'pollframes' object has no attribute '_enabled_devices'

MaryBD, 2019-02-26:
Is any python sample code that can simply run two cameras and capture their frames (seperately or merging)?

darrell-rg, 2019-02-26:
The PR I linked is not the main code, it is just a patch for the error you originally posted.  You still need to run "python box_dimensioner_multicam_demo.py" to use the main code,  but the poll_frames() function in realsense_device_manager.py must be fixed as shown in the PR.

MaryBD, 2019-02-27:
Thanks. Yes, as I mentioned I replaced it with the original poll_frames function and call it from main  realsense_device_manager.py  but the error is exactly as same as when I use the original codes:

device.pipeline.poll_for_frames(frameset)
TypeError: poll_for_frames(): incompatible function arguments. The following argument types are supported:
    1. (self: pyrealsense2.pyrealsense2.pipeline) -> pyrealsense2.pyrealsense2.composite_frame

Invoked with: <pyrealsense2.pyrealsense2.pipeline object at 0x0000018A3AD8BEA0>, <pyrealsense2.pyrealsense2.composite_frame object at 0x0000018A3AD92068>

 I print out the calibration chessboard but it never reaches that line!

darrell-rg, 2019-02-27:
You might be using an older version of the pyrealsense2 lib.  The lib you get from the installer or pip install seems to have problems or be out of date.     Try building the python lib from source,  there are some instructions here: 

https://github.com/IntelRealSense/librealsense/tree/development/wrappers/python

MaryBD, 2019-02-28:
Still doesn't work.

darrell-rg, 2019-03-01:
From the error message it looks like you are still using the old code.   Make sure your realsense_device_manager.py is this exact code, no other files should be changed: https://raw.githubusercontent.com/IntelRealSense/librealsense/44ac0c508fd8452005dee61e05c51ed15570045b/wrappers/python/examples/box_dimensioner_multicam/realsense_device_manager.py

MaryBD, 2019-03-04:
@darrell-rg Thank you, it works well now!

RealSenseCustomerSupport, 2019-08-13:
----------------------------------------------

Hi @MaryBD,

I will close this if no further concern.

closed 2019-09-06 by RealSenseCustomerSupport