realsenseai/librealsense #3326: Running Multicameras

MaryBD, 2019-02-25:
Hello, is there any exmaple on using multicams? I need to run two realsense D435 at the same time and have access to both frames and depth images of both cameras separately. thanks.

MartyG-RealSense, 2019-02-25:
Intel's webinar session on YouTube from 2018 about using multiple RealSense 400 Series cameras is an excellent introduction to multicam setups.

https://www.youtube.com/watch?v=drBCxHhbxI0

Also, the discussion linked to below has a lot of example scripting about controlling separate cameras in a multicam setup.

https://github.com/IntelRealSense/librealsense/issues/2219

MaryBD, 2019-02-26:
Thank you very useful video. Is there any sample code in python 3?

MartyG-RealSense, 2019-02-26:
Below is a user-contributed script for connecting two D415 cameras.  It should be adaptable for D435, though bear in mind that D435 can only sync depth streams and not color streams.  

https://github.com/IntelRealSense/librealsense/issues/1735#issuecomment-390840337

RealSenseCustomerSupport, 2019-03-04:
----------------------------------------------

@MaryBD Did you get through? Any other questions about this ticket? Looking forward to your update. Thanks!

MaryBD, 2019-03-04:
@MartyG-RealSense Thank you it works!

closed 2019-03-04 by MaryBD