realsenseai/librealsense #3327: modprobe: ERROR: could not insert 'videodev' using patch-realsense-ubuntu-lts.sh

XXLbot, 2019-02-25:
Hello Intel RealSense support team !

**Required Info**
Camera Model                                 D435
Firmware Version                             5.11.1.0
Operating System & Version            Ubuntu MATE 16.04.2 LTS
Kernel Version (Linux Only)               4.4.38-v7+                              
Platform                                           Raspberry Pi 3B
SDK Version                                     2.17.1
Language                                        (C++ and Python)
Segment                                         Robot/ROS

**Problem description**
I'm trying to install RealSense SDK 2.17.1 from source
(I got some errors with 2.18.0 on a basic PC/Windows10 and no error with 2.17.1)
I get a problem running the script ./scripts/patch-realsense-ubuntu-lts.sh

1) I ran 'sudo apt-get install  raspberrypi-kernel-headers' to avoid error
cp: cannot stat '/usr/src/linux-headers-4.4.38-v7+/.config': No such file or directory

2) I blacklisted module bcm2835_v4l2 to avoid error
Replacing videodev :
        Module videodev  is used by bcm2835_v4l2
        Unloading dependency bcm2835_v4l2
        ./scripts/patch-utils.sh: line 86: 18340 Segmentation fault      sudo modprobe -r ${unload_module_name}
Failed to unload module bcm2835_v4l2. error type 139 . Operation is aborted

3) I commented 4 lines in file scripts/patch-realsense-ubuntu-lts.sh to avoid error
cp: cannot stat '/home/eric/temp/librealsense-2.17.1/ubuntu-xenial-master/drivers/iio/accel/hid-sensor-accel-3d.ko': No such file or directory
Same solution as Issue #1743

4) Now when I run the script  ./scripts/patch-realsense-ubuntu-lts.sh I get the error
(All questions answered with [ENTER])
...
linux-headers-generic is already the newest version (4.4.0.142.148).
… 
From git://kernel.ubuntu.com/ubuntu/ubuntu-xenial
 * branch            master     -> FETCH_HEAD
Resetting local changes in ubuntu-xenial-master folder
 HEAD is now at be36faf UBUNTU: Ubuntu-4.4.0-142.168
Applying realsense-uvc patch
…
Applying realsense-metadata patch
...
Compiling uvc module
Compiling accelerometer and gyro modules
Compiling v4l2-core modules
**Patched kernels modules were created successfully**
… 
Replacing videodev :
Module is resident, unloading ...  succeeded.
Applying the patched module ... modprobe: ERROR: could not insert 'videodev': Exec format error
Failed to insert the patched module. Operation is aborted, the original module is restored
Verify that the current kernel version is aligned to the patched module version
The original  videodev  module was reloaded

$ ls /usr/src/
linux-headers-4.4.0-142  linux-headers-4.4.0-142-generic  linux-headers-4.4.38+  linux-headers-4.4.38-v7+
$
$dmesg | tail -n 10
[   32.506570] Bluetooth: BNEP socket layer initialized
[   32.542909] Bluetooth: RFCOMM TTY layer initialized
[   32.542938] Bluetooth: RFCOMM socket layer initialized
[   32.542953] Bluetooth: RFCOMM ver 1.11
[  752.021835] media: Linux media interface: v0.10
[  752.022446] videodev: version magic '4.4.38-v7+ SMP mod_unload modversions ARMv7 p2v8 ' should be '4.4.38-v7+ SMP mod_unload modversions ARMv7 '
[  752.118199] Linux video capture interface: v2.00
[ 1377.316373] media: Linux media interface: v0.10
[ 1377.317965] videodev: version magic '4.4.38-v7+ SMP mod_unload modversions ARMv7 p2v8 ' should be '4.4.38-v7+ SMP mod_unload modversions ARMv7 '
[ 1377.710100] Linux video capture interface: v2.00
$
Full log attached 
[patch-realsense-ubuntu-lts.sh__log-v1-2019.02.25.txt](https://github.com/IntelRealSense/librealsense/files/2901380/patch-realsense-ubuntu-lts.sh__log-v1-2019.02.25.txt)

Where comes this 'p2v8' from ??!!
What must be done in order to solve this problem and get the script running successfully ?
Thanks in advance for your response/help
Regards

ev-mp, 2019-02-26:
@XXLbot , the script you're trying to modify is designed and customized for Ubuntu LTS distributions.
Switching to Ubuntu MATE would require extensive modifications, so good for you on the most tedious part!
> Patched kernels modules were created successfully

As for loading the modules - one thing I can point is that there is a way to load kernel module even though the "vermagic" attribute is not identical - but you should be cautious about it as it can result in kernel panic or worse.
Please read [modprobe man](https://www.systutorials.com/docs/linux/man/8-modprobe/) to understand the risks before applying:
> sudo modprobe <module_name>.ko **--force-vermagic**

XXLbot, 2019-02-26:
First of all thanks for this response
Sorry for the dumb question
modeprobe expects module videodev.ko  to be located in  
/lib/modules/4.4.38-v7+/kernel/drivers/media/v4l2-core/
the module videodev.ko built by the script patch-realsense-ubuntu-lts.sh is located in  
/home/eric/temp/librealsense-2.17.1/ubuntu-xenial-master/drivers/media/v4l2-core/videodev.ko
what command(s) must be issued ?

Full path can be specified with insmod command ...
sudo insmod /home/eric/temp/librealsense-2.17.1/ubuntu-xenial-master/drivers/media/v4l2-core/videodev.ko
but get an error
insmod: ERROR: could not insert module /home/eric/temp/librealsense-2.17.1/ubuntu-xenial-master/drivers/media/v4l2-core/videodev.ko: Invalid module format
insmod doesn't support option  '--force-vermagic'
sudo insmod --force-vermagic  /home/eric/temp/librealsense-2.17.1/ubuntu-xenial-master/drivers/media/v4l2-core/videodev.ko
insmod: unrecognized option '--force-vermagic'
what must be done ?

Same problem with option -f
sudo insmod -f  /home/eric/temp/librealsense-2.17.1/ubuntu-xenial-master/drivers/media/v4l2-core/videodev.ko
insmod: ERROR: could not insert module /home/eric/temp/librealsense-2.17.1/ubuntu-xenial-master/drivers/media/v4l2-core/videodev.ko: Invalid module format
''insmod
 -f, --force
              Attempt  load  the  module  even  if  the version of the running
              kernel and the version of the kernel for which  the  module  was
              compiled  do  not match.  This only overrides the kernel version
              check, it has no effect on symbol name checks.   If  the  symbol
              names in the module do not match the kernel then there is no way''

ev-mp, 2019-02-26:
When you browse the `patch-realsense-ubuntu-lts.sh` you'll find the following segment:
>#Copy the patched modules to a sane location
sudo cp $KBASE/drivers/media/usb/uvc/uvcvideo.ko ~/$LINUX_BRANCH-uvcvideo.ko
...
echo -e "\e[32mPatched kernels modules were created successfully\n\e[0m"
#Load the newly-built modules
**try_module_insert** videodev	~/$LINUX_BRANCH-videodev.ko 	/lib/modules/`uname -r`/kernel/drivers/media/v4l2-core/videodev.ko

`try_module_insert` is a function that is defined in [patch_utils.sh](https://github.com/IntelRealSense/librealsense/blob/master/scripts/patch-utils.sh) where we'll copy the patched kernel module from the local "sandbox" to the location recognized by Ubuntu. Additionally it will backup the origin module in case the operation fails.

As for `'--force-vermagic'`  - you're trying it with `insmod` instead of `modprobe` for which it is defined.

XXLbot, 2019-02-26:
just an extra question....
I understand that after script is applyed the following command
modinfo uvcvideo | grep version:
should return a string like this one 'version: 1.1.1.realsense-4.4.0'
but
modinfo /home/eric/temp/librealsense-2.17.1/ubuntu-xenial-master/drivers/media/usb/uvc/uvcvideo.ko |  grep version:
only returns
version:        1.1.1
srcversion:     FF29E0FCAAF2138D1AB635A

is it a (potential) problem ?!

ev-mp, 2019-02-26:
It is not. The `realsense` suffix is present only in DKMS-distribution to cope with a situation of conflicting DKMS packages. It is normal not to see it when patching manually.

XXLbot, 2019-02-26:
@ev-mp Thanks for your help

1) As I mentioned earlier I tried `'--force-vermagic'`  with BOTH commands modprobe and insmod
modprobe is expecting modules in /lib/modules where the good/standard videodev.ko modules stays
                 so of course I get an error when I specify videodev.ko build by the Intel script in the temporary space

2) thanks for clarification about `realsense` suffix present only in DKMS-distribution

3) First you suggested (in order to avoid vermagic verification) to use modprobe --force-vermagic but how ?
Now you introduce 'try_module_insert' 
Do I have to use it for both  videodev.ko  and  uvcvideo.ko  modules ?
How should I use it out of any Intel script ? 

Regards

ev-mp, 2019-02-26:
Please see inline
>3\. First you suggested (in order to avoid vermagic verification) to use modprobe --force-vermagic but how ?
>3\. Now you introduce 'try_module_insert'

Imo you misinterpret 'try_module_insert' - this function is part of the patching script. It is not a linux built-in command.
It is used as a convenient aggregation of several coherent operations required to replace a working kernel module with a patched version.
Inside that function/script you'll find the module being copied to the standard location recognized by Ubuntu, so that `modprobe` would be able to pick and use it.
You may also copy its content and activate it directly in the modified version of the script that you maintain for Ubuntu Mate.

>3\. Do I have to use it for both  videodev.ko  and  uvcvideo.ko  modules ?

Yes. In the reference  `patch-realsense-ubuntu-lts.sh` we apply this procedure for all (4) modules patched by Librealsense

XXLbot, 2019-02-27:
Thanks for the update.
1) I'm going to have a look at  'try_module_insert'

2) In the mean time, 
at https://github.com/IntelRealSense/librealsense/blob/master/doc/RaspberryPi3.md
**Ubuntu MATE is clearly mentioned and was used/tested**
a) ''1. Install Ubuntu MATE on your Raspberry PI 3.'' 
b) on the 2nd photo
so could you tell me **which Ubuntu image was used/tested** by the Intel team in order **to write this tuto** (download URL and MD5/SHA) ?
Is it ubuntu-mate-16.04.2-desktop-armhf-raspberry-pi.img.xz
from https://ubuntu-mate.org/download/
with SHA256 dc3afcad68a5de3ba683dc30d2093a3b5b3cd6b2c16c0b5de8d50fede78f75c2

I'm using this image and I'm astonished to face so many ''problems'' not described in the tuto...

By the way, did anybody test the image ubuntu-16.04-preinstalled-server-armhf+raspi3.img.xz
located at https://ubuntu-pi-flavour-maker.org/download/
Ubuntu Classic Server 16.04 Raspberry Pi 3 (216MB)

Regards

ev-mp, 2019-02-27:
@XXLbot, 
RaspberryPi with Ubuntu Mate should provide the basic depth/ir streaming and controls functionality without the need for kernel patches. 
This is correct with most Linux distributions with kernels that include the upstreamed depth format (4.4.60+, iirc) and also [ the way we recommend to proceed](https://github.com/IntelRealSense/librealsense/blob/master/doc/RaspberryPi3.md#installation-instructions) in the installation:
>In the section Prerequisites, proceed with the steps till (not including) the kernel patches instructions.

I'd suggest to finish the basic installation first according to guide to verify that streaming works.
Once accomplished, you'll have ~90% of D435 functionalities running. Then you'll be able to assess whether porting the patches is needed.

P.S: I'll also try to obtain the exact distro version used in the tutorial.

XXLbot, 2019-03-03:
OK, I continued installation on Friday, I will update on Monday.

XXLbot, 2019-03-04:
Hello

1) I finished the installation according to the guide
cd ~/temp/librealsense-2.17.1/ 
mkdir build && cd build
cmake ../ -DBUILD_EXAMPLES=true 
sudo make uninstall && make clean && make -j4 && sudo make install  

When I start realsense-viewer the application opens with the title bar ''Intel RealSense D435 [logoUSB] 2.1'' 
I get the following messages

in Output window
librealsense version: 2.17.1
Intel RealSense D435 was selected as a default device

in the Terminal where realsense-viewer is started
WARNING [] (sensor.cpp:338) Unregistered Media formats : [ UYVY ]; Supported: [ ] 
I get the SAME MESSAGE on an official/supported PC Intel Core i7 / Ubuntu 16.04
**Question#1 : What does this message mean ? Is it important ? How to fix/avoid it ?**

RGB Camera Off→On	First window opens
WARNING [] (ds5-timestamp.cpp:64) UVC metadata payloads not available. Please refer to installation chapter for details.
Stereo Module Off→On	Second window opens
WARNING [] (ds5-timestamp.cpp:64) UVC metadata payloads not available. Please refer to installation chapter for details.
**Question#2 : The target is to use RealSense ROS Wrapper. 
Will PointCloud be available with such message, and I assume, missing feature ?**

2) Coming back to patch-realsense-ubuntu-lts.sh
during the script's execussion I get the message 
HEAD is now at be36faf UBUNTU: Ubuntu-4.4.0-142.168
The processor is Raspberry Pi 3B, raspberrypi-kernel-headers is installed but not mentioned in the log. 
**Is it a potential problem ?** 

Thanks in advance for your responses
Regards

XXLbot, 2019-03-05:
Hi ev-mp  Any update ?

XXLbot, 2019-03-05:
Hi [https://github.com/ev-mp](url)  Any update ?

XXLbot, 2019-03-06:
I followed comment in librealsense2 issue #1132
and choosed to build the SDK from source with CMake flag -DFORCE_LIBUVC=true. 

Old test (without -DFORCE_LIBUVC=true), I got 
realsense-viewer
WARNING [] (sensor.cpp:338) Unregistered Media formats : [ UYVY ]; Supported: [ ]
New test (with -DFORCE_LIBUVC=true), I get 
realsense-viewer
WARNING [] (sensor.cpp:338) Unregistered Media formats : [ P UYVY Y8   ]; Supported: [ ]

Now using ROS I only get one **librealsense2** message
roslaunch realsense2_camera rs_camera.launch  enable_color:=false enable_infra1:=false  enable_infra2:=false enable_imu:=false enable_fisheye:=false  enable_depth:=true depth_width:=480  depth_height:=270  depth_fps:=6 enable_pointcloud:=true
WARNING [] (sensor.cpp:338) Unregistered Media formats : [ P UYVY Y8   ]; Supported: [ ]

**Question#1** 
How can we fix this warning message?

**Question#2a** 
Any news about Ubuntu MATE 16.04 image used by Intel team to write the RealSense/RaspberryPi3 tuto at https://github.com/IntelRealSense/librealsense/blob/master/doc/RaspberryPi3.md ?
URL, MD5/SHA, Kernel version,  PreReq, …
**We are still interested by running the script because ''-DFORCE_LIBUVC=true'' workaround option seems quite unstable.....**

**Question#2b**
What about the message ''HEAD is now at be36faf UBUNTU: Ubuntu-4.4.0-142.168 ''
during the script patch-realsense-ubuntu-lts.sh execution ?

Thanks in advance for your help,
Regards

ev-mp, 2019-03-06:
Hello @XXLbot , please see some andwers below:
>WARNING [] (sensor.cpp:338) Unregistered Media formats : [ UYVY ]; Supported: [ ]
I get the SAME MESSAGE on an official/supported PC Intel Core i7 / Ubuntu 16.04
Question#1 : What does this message mean ? Is it important ? How to fix/avoid it ?

This warning does not affect the device functionality and can be disregarded. (UYVY  is the format for synthetic color from Left imager supported with D415)

>WARNING [] (ds5-timestamp.cpp:64) UVC metadata payloads not available. Please refer to installation chapter for details.
Question#2 : The target is to use RealSense ROS Wrapper.
Will PointCloud be available with such message, and I assume, missing feature ?

Metadata requires the kernel patch to be properly applied when using v4l kernel module. This is in the extra 10% I mentioned above. The feature is elaborated in this [link](https://github.com/IntelRealSense/librealsense/blob/master/doc/frame_metadata.md#metadata-support-for-intel-realsense-devices) .
Pointcloud functionality is provided regardless of metadata on/off, though enabling it should make the results more robust in terms of depth-color synchronization.

>WARNING [] (sensor.cpp:338) Unregistered Media formats : [ P UYVY Y8 ]; Supported: [ ]
Question#1
How can we fix this warning message?

Can be disregarded - both with libuvc and v4l kernel modules on linux you should be able to stream Depth and IR streams. Applying the kernel patches (given that you switch to video4linux backend) would reduce the list to `[ UYVY ];` , but will not add to functionality.

>Question#2a
Any news about Ubuntu MATE 16.04 image

No updates yet.
The [documentation](https://github.com/IntelRealSense/librealsense/blob/master/doc/RaspberryPi3.md) provides a snapshot with the Mate version. 
[image]
Does it help?

>Question#2b
What about the message ''HEAD is now at be36faf UBUNTU: Ubuntu-4.4.0-142.168 ''
during the script patch-realsense-ubuntu-lts.sh execution ?

As stated above, the script is for Ubuntu LTS releases, and the message reflects the latest version of the LTS kernel posted by Ubuntu team that the script downloads.  As of now we do not provide kernel patches port for Ubuntu Mate.

XXLbot, 2019-03-06:
@ev-mp First of all thanks for your response, I realy appreciate your help

Regarding the information you posted  ''MATE version 1.16.2 '' and your question ''Does it help?''
unfortunatelly not. 
I understand that Ubuntu MATE 16.04.2 LTS is an (unofficial) version of Ubuntu 16.04.2 LTS with MATE desktop environment preinstalled so my question about Ubuntu 16.04 LTS version used by Intel Team to test RealSense D400 is still needed.
With such information in hand I should be able to properly apply patch-realsense-ubuntu-lts.sh without any problem and 1) get videodev and uvcvideo kernel modules properly loaded 2) get a more stable D435 environment 3) get 100% of D435 features/functionalities.
The only Ubuntu 16.04 LTS for ARM plateform (ARMhf) supported by Ubuntu is a server image.

I still do not understand why in https://github.com/IntelRealSense/librealsense/blob/master/doc/RaspberryPi3.md
it is asked not to apply the kernel patch script ''In the section Prerequisites, proceed with the steps till (not including) the kernel patches instructions.'' because without kernel patch we get MetaData/UVC  messages and not 100% of the D435 features. Any comments ?

BTW I should investigate other ARM plateform that offers USB3 (Odroid XU4, …) but D435 working with Raspberry Pi 3B in a stable way is mandatory for us...

Regards

ev-mp, 2019-03-06:
I think it is my misunderstanding as I though you were looking for the Ubuntu Mate version used in the guide.
The LTS versions used for verification can be downloaded from [the official release page](http://releases.ubuntu.com/16.04/) -
v16.04.1 and v16.04.6 are the most common, as well as  [v18.04](http://releases.ubuntu.com/18.04/), All AMD64.
For instance:
>Description:	Ubuntu 16.04.6 LTS
Release:	16.04
Codename:	xenial

[image]

XXLbot, 2019-03-07:
Hello @ev-mp
You were right I would like to know which Ubuntu 16.04 version was used by Intel team to write the Raspberry Pi 3B tuto 
I suspect Ubuntu MATE 16.04 LTS was used (stated by the screen shot) 
but maybe the Desktop MATE environment was installed on top of an other Ubuntu 16.04 LTS image. Please let me know.
Regards

ev-mp, 2019-03-07:
Hello @XXLbot, 
The guide was written with Ubuntu MATE 16.04 (ARM). 
But as stated before - there are not patches for this distro as its kernel is different from the LTS series for which Intel provides official support.
Once Ubuntu LTS is extended to ARM platforms, then the maintained patches should apply automatically. 

We are engaged in patches up-streaming, but at the same time we're committed to continue kernel support for Ubuntu LTS.
Note that even though there is no Intel official support for RaspberryPi, as well as many other platforms - Odroid/JetsonTX/etc', we will try to assist you addressing technical issues and providing support where appropriate.

XXLbot, 2019-03-08:
Hello ev-mp
Thanks for your update and clarification about version used for the Raspberry Pi 3B guide
I close this issue
Regards

closed 2019-03-08 by XXLbot