realsenseai/librealsense #3328: Failed to run examples on Android development board

yichenpan, 2019-02-25:
| Camera Model                        | D435 | 
| Firmware Version                    | v5.11.1 | 
| Operating System & Version | Andriod O | 
| Platform                                  | HiKey 960 Board |
| SDK Version                            |v2.18.0| 
| Language                                | C | 
| Segment			         | Andriod | 

### Issue Description

Hi All,
I was trying to build and run librealsense on the reference board recommended by ASOP: [HiKey 960](https://source.android.com/setup/build/devices).

I successfully built examples on my ubuntu machine and copied them to the board through adb by following the [instructions](https://github.com/IntelRealSense/librealsense/blob/master/doc/android/AndroidNativeSamples.md).

However, I encountered an issue when I was trying to run the examples. Since I was not able to run as superuser in the terminal emulator on this board and I need to use the OTG for connecting my sensor, I had to run the examples via adb-over-WiFi. Then I saw the following error message:

```
There are 1 connected RealSense devices.
rs_error was raised when calling rs2_create_device(info_list:0x7fb7717e00, index:0)
       Could not open device. Last Error: No route to host
```

It looks to me like a permission issue of the USB port, but I was running with superuser permission. I'm wondering if anyone has an idea how to solve it? Or encountered the same issue? Thanks in advance.

dorodnic, 2019-02-26:
Hi @yichenpan 
Please note that we have just re-engineered the whole Android backend to avoid the need for superuser permissions. [github.com/IntelRealSense/librealsense/blob/**development**/doc/android.md](https://github.com/IntelRealSense/librealsense/blob/development/doc/android.md)
It is still being worked on, but already shows much better potential in terms of stability and scalability. Please consider giving it a try. 
I personally never encountered `Last Error: No route to host`, even when running adb via Wifi. Maybe someone else is familiar with it

yichenpan, 2019-02-26:
Hi @dorodnic 
Thanks for your reply. I sure will give the development backend a try.

As for the error message part, after investigation, I believe it's due to the missing UVC driver on this board's native Andriod. I will try to recompile the kernel with USB video configuration and share my result here.

RealSenseCustomerSupport, 2019-03-04:
----------------------------------------------

@yichenpan Did you get chance to try the development branch?Any good news? Looking forward to your update. Thanks!

RealSenseCustomerSupport, 2019-03-07:
----------------------------------------------

@yichenpan Now the unroot android version is included in 2.19 librealsense. Could you please have a try and update your test result? Thanks!

yichenpan, 2019-03-08:
@RealSenseCustomerSupport Hi, sorry for the late reply. I just gave it a try today. Results are the following:
**Camera App:** Can be successfully run on the board, but the RGB streaming stops after first few seconds while the depth streaming seems to work properly.
[image]
**Capture App:** Crash everytime and the whole device just stop responding.

RealSenseCustomerSupport, 2019-03-08:
----------------------------------------------

@yichenpan
1. Camera App: You can click the wrench button at the bottom left to configure the streaming. Could you please have a try if you still see the RGB streaming stuck issue when only RGB streaming enabled?

2. Capture App: Did you clear/kill or stop Camera App before you start Capture App? If you still get crash, please help collect adb logcat logs. Thanks!

RealSenseCustomerSupport, 2019-03-11:
----------------------------------------------

@yichenpan Any update? Do you still have issue for Capture App?

yichenpan, 2019-03-11:
@RealSenseCustomerSupport Hi, sorry for the late reply. I don't have the access for the board during the weekend. The streaming stuck issue of Camera app has been fixed with your solutions. However, Capture app is still not working... I have attached the log here for your reference.
[log.txt](https://github.com/IntelRealSense/librealsense/files/2953921/log.txt)

RealSenseCustomerSupport, 2019-03-20:
----------------------------------------------

@yichenpan For capture App issue you met, not very sure if it's the same as what I saw here. I need start App twice to make it working. For the first start, the image is not showing on GUI. Then clear/stop the APP and restart the capture APP, this time the depth and color image shown.  Could you please have a try to start APP twice to see if it works for you?

yichenpan, 2019-03-20:
@RealSenseCustomerSupport I think we're having different issues here... My capture app crashes every time, no matter how many times I tried, while the camera app has no such issue at all.

RealSenseCustomerSupport, 2019-03-21:
----------------------------------------------

@yichenpan Did you change the configurations in the sample code? Noticed in the logcat you posted, color and depth in fact start streaming there. But Depth is 90fps which might cause USB bandwidth problem. If you changed the code, could you please try 640x480@30fps first?

03-11 19:27:51.620  2295  2306 I ActivityManager: START u0 {act=android.intent.action.MAIN cat=[android.intent.category.LAUNCHER] flg=0x10200000 cmp=com.intel.realsense.capture/.MainActivity (has extras)} from uid 10010 on display 0
03-11 19:28:01.625  2295  2309 W ActivityManager: Launch timeout has expired, giving up wake lock!
--------- beginning of main
03-11 19:29:48.679  3524  3553 D librs  : DISPATCH (TS: )--> Color 3465 139561.807000
03-11 19:29:48.679  3524  3553 D librs  : DISPATCH (TS: (FN: I Depth I Infrared I Infrared ) I Color )--> Color 3465 139561.807000
03-11 19:29:48.679  3524  3553 D librs  : I Color--> Color 3465, 139561.807000
03-11 19:29:48.679  3524  3553 D librs  : SYNC (TS: (FN: I Depth I Infrared I Infrared ) I Color )--> Color 3465 139561.807000
03-11 19:29:48.679  3524  3553 D librs  : fps 30 Color 3465 139561.807000
03-11 19:29:48.680  3524  3553 D librs  : (TS: (FN: I Depth I Infrared I Infrared ) I Color )Color 3465 139561.807000 fps 30 gap 33.3333 next_expected: 139595
03-11 19:29:48.680  3524  3553 D librs  : fps 30 Color 3465 139561.807000
03-11 19:29:48.680  3524  3553 D librs  : **fps 90 Depth** 95 139523.406000

RealSenseCustomerSupport, 2019-03-26:
----------------------------------------------

@yichenpan  Any update?

RealSenseCustomerSupport, 2019-04-01:
----------------------------------------------

@yichenpan Did you try 640x480@30fps? Any issue there? Thanks!

RealSenseCustomerSupport, 2019-04-04:
----------------------------------------------

@yichenpan If no other questions about this ticket, will close it accordingly.

closed 2019-04-11 by RealSenseCustomerSupport