realsenseai/librealsense #3329: realsense-viewer fails after force-quit

whg-res, 2019-02-26:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 5.10.06 | 
| Operating System & Version | Ubuntu 16.04  | 
| Kernel Version (Linux Only)    |  4.4.0 | 
| Platform                                 | PC |
| SDK Version                            |  2.16.1  | 

### Issue Description
I'm checking stability of RealSense camera and I find realsense-viewer 
behaves under specific condition. I want to know this is realsense-viewer bug or other problem.

#### how to reproduce 
- Open realsense-viewer
- Set "Auto Exposure Priority = OFF"
- Set all of Depth, Infrared 1 and Infrared 2 check boxes ON
- Start Stereo Module stream
- Start RGB Camera stream
- Kill realsense-viewer by Ctrl+C
- Open realsense-viewer again
- Start Stereo Module stream

These operations result in "IR stream start sailure" and "No Frames Received" error.
Even if I try the same process with other FW(5.11.1) and SDK(2.18.1) the behavior is the same.
And also, when I start RGB Camera stream first and start Stereo Module stream next,
the errors do not emerge.

Is it a bug in realsense-viewer? If so, is there any workaround when I make software with SDK?

MartyG-RealSense, 2019-02-26:
If a camera stream is not properly shut down then it may result in problems such as the 'frames didn't arrive' error.  This is a known issue with the firmware that has been documented in the SDK release notes as recently as version 2.17.0.  Details of this error can be found in the link below.

https://github.com/IntelRealSense/librealsense/issues/1086

dorodnic, 2019-02-26:
Yep, that's a known issue unfortunately... It is being investigated (for some time now) and firmware and hardware level. 
For now there are several work-arounds. I believe @doronhi has recently added Ctrl+C trap to the ROS node to work-around this problem, perhaps we need to do the same in the Viewer. Also, you could call Hardware Reset command to re-initialize the camera or simply re-connect it if that's an option.

whg-res, 2019-02-26:
Thank you for good reference. I couldn't find it.
Unfortunately, re-connecting the device is not practical solution for my system. But calling hardware reset should be fine. Does it simply mean calling dev.hardware_reset() in my software?

MartyG-RealSense, 2019-02-26:
hardware_reset should be placed near the start of your script, like in the example provided by Dorodnic in the link below.

https://github.com/IntelRealSense/librealsense/issues/2161#issuecomment-408779150

whg-res, 2019-02-27:
Thank you for your kind instruction. I will follow that.

closed 2019-02-27 by whg-res

SamLewis0602, 2019-03-21:
@dorodnic Hi Sergey, I am looking to use the following code as you suggested in #2161:

```
rs2::context ctx;
rs2::device dev = ctx.query_devices().front(); // Reset the first device
dev.hardware_reset();
rs2::device_hub hub(ctx);
dev = hub.wait_for_device();
```

However, we are using the python wrapper, so I modified it to this:
```
ctx = rs.context()
devices = ctx.query_devices()
for dev in devices:
        dev.hardware_reset()
```

I don't see a device hub option to wait for the device to be ready again with via the Intellisense in my IDE, however I do see device_list. Can you give an example of the device_hub in the python wrapper? Thanks in advance for any help.

AndreV84, 2021-10-01:
@MartyG-RealSense 
Is it possible to force reset of realsense cameras somehow in case the python script with hardware reset won't work?
but after reboot of PC device comes up?

MartyG-RealSense, 2021-10-01:
@AndreV84   As described in https://github.com/IntelRealSense/librealsense/issues/8393 you could run a Linux bash script that resets the entire USB port instead of just the camera.

AndreV84, 2021-10-02:
@MartyG-RealSense 
Thank you for pointing out!
However could you extend which exactly bash script, please? I was not able to find any at the https://github.com/IntelRealSense/librealsense/issues/8393

AndreV84, 2021-10-02:
seems this one https://wiki.ubuntuusers.de/usbreset/
it is rather c executable than bash script though

AndreV84, 2021-10-02:
@MartyG-RealSense 
However, the problem is that after the camera disappear [ after running software] it is not at the specified port anymore
```
 sudo ./usbreset /dev/bus/usb/007/006
Error opening output file: No such file or directory

```
However before running the software it would be on the usb outputs also would seem to return successful reset.
Any ideas?

MartyG-RealSense, 2021-10-02:
Another RealSense user in https://github.com/IntelRealSense/librealsense/issues/7060#issuecomment-671396923 who used a usb reset script experienced a **No such device** error too after the reset.  A solution was not reached in that particular case.

Another user had success using a Python reset script called **PyUSB** and said that they were able to reset the camera with it as often as they wanted.

https://github.com/IntelRealSense/librealsense/issues/8274#issuecomment-770191698
https://github.com/pyusb/pyusb