realsenseai/librealsense #3330: Floor Detection

chetansarmak, 2019-02-26:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   | 05.11.01.00 | 
| Operating System & Version |   Linux Ubuntu 16  | 
| Kernel Version (Linux Only)    |  4.4.0-21                                        | 
| Platform                                 | PC  |
| SDK Version                            | 4.4.0-21                          | 
| Language                            |  c++                          | 
| Segment			|  Robot                          | 

### Issue Description

How can I use the depth values from Realsense to do floor segmentation,any help is appreciated?
Thanks in advance!!

MartyG-RealSense, 2019-02-26:
If the camera is pointing directly downwards to the floor, an easy way to remove it may be to treat the floor like a background.  You could set a 'depth clamp' with a maximum depth value (MaxZ), or a 'Disparity Shift' value.  For example, the smaller a MaxZ value, the more of the floor 'background' that would disappear as the maximum distance that the camera can depth-scan is reduced.  Increasing the 'Disparity Shift' value has the same effect - the higher the value, the lower the maximum observable distance becomes.

The link below discusses this subject.

https://forums.intel.com/s/question/0D70P000006B7gpSAC

If the camera is pointing straight ahead and you want to segment the floor, that could be more challenging to achieve.

Ground plane segmentation with RealSense is discussed in the research paper linked to below.

https://www.ncbi.nlm.nih.gov/pmc/articles/PMC5134613/

chetansarmak, 2019-02-27:
Thank you so much Marty !!
That article is quite helpful and but they do not provide any source of their code 
which becomes tough to implement .
I could able to get the point cloud data , applying RANSAC algorithm like they mentioned is challenge for me  !!
Any lead on it?

MartyG-RealSense, 2019-02-27:
A RANSAC algorithm seems to be a common part of RGB-D based SLAM software.  

I located another ground segmentation paper that focuses strongly on RANSAC.

https://www.researchgate.net/publication/307572427_Fast_and_accurate_ground_plane_detection_for_the_visually_impaired_from_3D_organized_point_clouds

You can find further material by googling for 'slam ransac ground segmentation'.

A popular SLAM solution that can make use of RANSAC is **rgbdslam**

http://wiki.ros.org/rgbdslam

closed 2019-03-16 by RealSenseCustomerSupport