realsenseai/librealsense #3332: Automatic object detection

KarlAbson, 2019-02-26:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D415 / D435 } | 
| Firmware Version                   | (05.10.06) | 
| Operating System & Version |   {Win 10  | 
| Platform                                 | PC |
| SDK Version                            |  { 2.16.5 }                          | 
| Language                            |  {C# }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

I have a question in regards to object detection and as to how the D415s depth information can be used in this task. I'm currently creating a more advanced version of RS_Measure. This program will automatically find objects for you and then measure them automatically. 

At the moment I'm finding objects by converting the colour frame into a binary image and searching for contours using EMGU / Open CV. This works well but I'm wondering if there is away of utilizing the camera more in this task and using it's depth capabilities for this. Currently I am only using the depth feed for measuring the objects after they have been found. 

The objects are placed on a desk underneath the camera approximately 50/60cm away as shown in the image. Can anyone think of any interesting ways of utilizing the depth capabilities? Also is there away of evaluating the depth quality of the camera? I want to be able to tell the user if they have bad conditions (lighting, positioning) etc.

[image]

MartyG-RealSense, 2019-02-26:
An alternative means of object detection is to use a Deep Neural Network (DNN), utilizing OpenCV and RGB imagery.   There are a couple of example programs for the SDK that demonstrate this, recognizing objects and providing information about their distance in meters from the camera.

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/opencv/dnn

https://github.com/twMr7/rscvdnn

In regard to providing user feedback about depth quality, you could get the camera's current environmental values (gain, exposure, etc) from the frame metadata and then display messages based on the ranges that those values are currently in.

https://github.com/IntelRealSense/librealsense/blob/master/doc/frame_metadata.md

It would be up to you to decide which value ranges constitute what is bad and what is good.

In regard to positioning, you could perhaps generate a bounding box with OpenCV and provide feedback if a detected object is not fully within the box.

https://stackoverflow.com/questions/14505679/drawing-bounding-box-using-opencv-in-c-enviroment?language=en_US

KarlAbson, 2019-02-26:
Hallo there,

Many thanks for your suggestion. I appreciate you taking the time to help
out. Unfortunately it doesn't answer the question that I have in this case.
I'm not looking for another RGB approach. I'm already using these EMGU /
OpenCV methods with the RGB camera. My question is regarding the depth
sensors and making more use of them. It must be possible to utilize them
more in actually finding the objects. Currently they are only being used to
perform the measurement.

Unfortunalty I am looking for very small objects on a table. A lot of the
ideas I have had are only useful for larger objects. For instance the
colonization images are not much use as the objects appear the same colour
as the table.

The idea regarding using the environmental values is good. The exposure
would be good for saying if the area is too bright. I wonder if there are
possibilities for identifying if the camera is tilted incorrectly. It
should be aimed correctly right down at the table for instance and not at
an angle.

MartyG-RealSense, 2019-02-26:
There have been a few cases where depth sensing has been used as a means to detect obstacles for visually impaired people.  Like the application called 'sonic-sight'.

https://forums.intel.com/s/question/0D50P00004AszxySAB/do-you-have-a-blind-friend-d435-some-time-and-would-wish-to-test-an-echolocation-app-otherwise-share-your-opinion?language=en_US

closed 2019-03-17 by RealSenseCustomerSupport