realsenseai/librealsense #3333: RealSense D435 disconnects

elvyfeliz, 2019-02-26:
Hello All,

We are experiencing a lot of issues with the RealSense D435 where it disconnects and crashes our application. Sometimes the computer doesn't recognize that it's connected. Sometimes rebooting the computer fixes the problem and sometimes it doesnt. Sometimes we have to unplug and plug back.

Here's our environment where we are experiencing the issue,

**Intel NUC I3**
BIOS Information
        Vendor: Intel Corp.
        Version: JYGLKCPX.86A.0031.2018.0301.1357

**Realsense D435**
   Firmware: 5.9.02

**Ubuntu 16.04**
4.13.0-1024

Here's the first issue that I noticed. 

1. I have a few cameras here, some of them when you connect them to the computer and look at the dmesg or syslog this is what you get in the logs

`Product: Intel(R) RealSense(TM) 430
`
- This works fine okay with our application and realsense-viewer.

However, we have some cameras with the same model (D435) that when you connect them in the computer this is what I get,

```
Product: Intel(R) RealSense(TM) SR400
SerialNumber: 111111111111111 
```

- This doesn't work at all with our application or the realsense-viewer application. It's not even recognize as a camera on the computer.
The serial number is weird, 111111... while it also mentions that it's a SR400 instead of 430.

2. We are experiencing a lot of problems with these cameras. To the point where we have to replace them, unplug them, reboot the machine, replace the NUC, replace cables, add usb hubs, etc. Still the issue persist. We need your help to identify where is the problem. Below are the error messages I see in the linux box.

Sometimes we get this error and we have to unplug it and plug it back.

```
[Mon Feb 25 15:36:38 2019] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).

[Mon Feb 25 15:36:38 2019] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).

[Mon Feb 25 15:36:38 2019] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024)
```

Sometimes when you reboot the machine and run lsusb it shows either 

```
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/8p, 10000M

    |__ Port 2: Dev 4, If 4, Class=Video, Driver=uvcvideo, 5000M

    |__ Port 2: Dev 4, If 2, Class=Video, Driver=uvcvideo, 5000M

    **|__ Port 2: Dev 4, If 0, Class=Video, Driver=, 5000M

    |__ Port 2: Dev 4, If 3, Class=Video, Driver=, 5000M**

    |__ Port 2: Dev 4, If 1, Class=Video, Driver=uvcvideo, 5000M
```

OR

```
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/7p, 5000M

    |__ Port 2: Dev 3, If 0, Class=Video, Driver=, 5000M

    |__ Port 2: Dev 3, If 1, Class=Video, Driver=, 5000M

    |__ Port 2: Dev 3, If 2, Class=Video, Driver=, 5000M

    |__ Port 2: Dev 3, If 3, Class=Video, Driver=, 5000M

    |__ Port 2: Dev 3, If 4, Class=Video, Driver=, 5000M
```

When the camera shows like the above it doesn't work in the realsense application or our application. Both application think there's no camera connected. To fix this we have to restart the computer or unplug and plug the camera back. Here's how it looks when the camera is good to go,

```
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/7p, 5000M

    |__ Port 2: Dev 4, If 0, Class=Video, Driver=uvcvideo, 5000M

    |__ Port 2: Dev 4, If 1, Class=Video, Driver=uvcvideo, 5000M

    |__ Port 2: Dev 4, If 2, Class=Video, Driver=uvcvideo, 5000M

    |__ Port 2: Dev 4, If 3, Class=Video, Driver=uvcvideo, 5000M

    |__ Port 2: Dev 4, If 4, Class=Video, Driver=uvcvideo, 5000M
```

As you can see, you can tell that it's bad when it's partially missing "Driver=uvcvideo" or completely missing "Driver=uvcvideo" when running lsusb.

Another error is the one below,
```
[ 1059.191080] media: media_devnode_register: cdev_device_add failed

[ 1059.199967] media: media_devnode_register: cdev_device_add failed
```

To fix this one we have to restart the camera as well.

There's also,

```
[Mon Feb 25 14:13:25 2019] usb 2-3.2: Device not responding to setup address.

[Mon Feb 25 14:13:25 2019] usb 2-3.2: Device not responding to setup address.

[Mon Feb 25 14:13:25 2019] usb 2-3.2: device not accepting address 84, error -71

[Mon Feb 25 14:13:25 2019] usb 2-3.2: Device not responding to setup address.

[Mon Feb 25 14:13:26 2019] usb 2-3.2: Device not responding to setup address.

[Mon Feb 25 14:13:26 2019] usb 2-3.2: device not accepting address 85, error -71

[Mon Feb 25 14:13:26 2019] usb 2-3.2: Device not responding to setup address.

[Mon Feb 25 14:13:26 2019] usb 2-3.2: Device not responding to setup address.

[Mon Feb 25 14:13:26 2019] usb 2-3.2: device not accepting address 86, error -71

[Mon Feb 25 14:13:26 2019] usb 2-3.2: Device not responding to setup address.

[Mon Feb 25 14:13:27 2019] usb 2-3.2: Device not responding to setup address.

[Mon Feb 25 14:13:27 2019] usb 2-3.2: device not accepting address 87, error -71

[Mon Feb 25 14:13:27 2019] usb 2-3-port2: unable to enumerate USB device
```

For the error above, I haven't find a way to fix a camera with this error.

3. Another thing I want to point out is that when I install the Realsense application using the following steps, https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md  the camera doesn't work and give's the following error,

```
25/02 17:22:10,842 WARNING [139829304096512] (backend-v4l2.cpp:1013) Frames didn't arrived within 5 seconds

 25/02 17:22:15,842 WARNING [139829304096512] (backend-v4l2.cpp:1013) Frames didn't arrived within 5 seconds

 25/02 17:22:20,846 WARNING [139829304096512] (backend-v4l2.cpp:1013) Frames didn't arrived within 5 seconds
```

I know is a lot. But we are experiencing a lot of issues. Can you guys help us out here and point us to the right direction?

dorodnic, 2019-02-26:
Hi @elvyfeliz 
First thing I **strongly** recommend is to perform a [firmware update](https://www.intel.com/content/www/us/en/support/articles/000028171/emerging-technologies/intel-realsense-technology.html). It might not solve all of the problems, but is very likely to help with some.

elvyfeliz, 2019-02-26:
Hey @dorodnic 

Which release version? I have updated to 5.10 before and I still experience daily disconnects.

RealSenseCustomerSupport, 2019-03-01:
----------------------------------------------

Hi @elvyfeliz

Please use FW 5.11.1
https://github.com/IntelRealSense/librealsense/releases

mikepurvis, 2019-03-04:
I can confirm that this problem remains with the 5.11.1.100 firmware. It seemed a lot better when I first installed it— the devices were fine across ~5 complete power cycles of the computer, but this morning they both came up dead again:

```
$ dmesg | grep usb
[    0.188055] usbcore: registered new interface driver usbfs
[    0.188063] usbcore: registered new interface driver hub
[    0.188089] usbcore: registered new device driver usb
[    1.556077] usb usb1: New USB device found, idVendor=1d6b, idProduct=0002
[    1.556079] usb usb1: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    1.556080] usb usb1: Product: EHCI Host Controller
[    1.556082] usb usb1: Manufacturer: Linux 4.15.0-45-generic ehci_hcd
[    1.556083] usb usb1: SerialNumber: 0000:00:1a.0
[    1.576073] usb usb2: New USB device found, idVendor=1d6b, idProduct=0002
[    1.576075] usb usb2: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    1.576076] usb usb2: Product: EHCI Host Controller
[    1.576078] usb usb2: Manufacturer: Linux 4.15.0-45-generic ehci_hcd
[    1.576079] usb usb2: SerialNumber: 0000:00:1d.0
[    1.577807] usb usb3: New USB device found, idVendor=1d6b, idProduct=0002
[    1.577809] usb usb3: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    1.577810] usb usb3: Product: xHCI Host Controller
[    1.577811] usb usb3: Manufacturer: Linux 4.15.0-45-generic xhci-hcd
[    1.577812] usb usb3: SerialNumber: 0000:00:14.0
[    1.578472] usb usb4: New USB device found, idVendor=1d6b, idProduct=0003
[    1.578473] usb usb4: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    1.578474] usb usb4: Product: xHCI Host Controller
[    1.578476] usb usb4: Manufacturer: Linux 4.15.0-45-generic xhci-hcd
[    1.578477] usb usb4: SerialNumber: 0000:00:14.0
[    1.892028] usb 1-1: new high-speed USB device number 2 using ehci-pci
[    1.912028] usb 2-1: new high-speed USB device number 2 using ehci-pci
[    1.915361] usb 4-3: Device not responding to setup address.
[    2.052438] usb 1-1: New USB device found, idVendor=8087, idProduct=0024
[    2.052440] usb 1-1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[    2.072417] usb 2-1: New USB device found, idVendor=8087, idProduct=0024
[    2.072419] usb 2-1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[    2.123337] usb 4-3: Device not responding to setup address.
[    2.340025] usb 1-1.2: new full-speed USB device number 3 using ehci-pci
[    2.392502] usb 4-3: device not accepting address 2, error -71
[    2.450160] usb 1-1.2: New USB device found, idVendor=8087, idProduct=07dc
[    2.450162] usb 1-1.2: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[    2.667355] usb 4-3: Device not responding to setup address.
[    2.875557] usb 4-3: Device not responding to setup address.
[    3.085306] usb 4-3: device not accepting address 3, error -71
[    3.120053] usb usb4-port3: attempt power cycle
[    4.059320] usb 4-3: Device not responding to setup address.
[    4.267632] usb 4-3: Device not responding to setup address.
[    4.484042] usb 4-3: device not accepting address 4, error -71
[    4.719614] usb 4-3: Device not responding to setup address.
[    4.939628] usb 4-3: Device not responding to setup address.
[    5.156042] usb 4-3: device not accepting address 5, error -71
[    5.192065] usb usb4-port3: unable to enumerate USB device
[    5.359625] usb 4-4: Device not responding to setup address.
[    5.579605] usb 4-4: Device not responding to setup address.
[    5.796042] usb 4-4: device not accepting address 6, error -71
[    6.031621] usb 4-4: Device not responding to setup address.
[    6.251619] usb 4-4: Device not responding to setup address.
[    6.468041] usb 4-4: device not accepting address 7, error -71
[    6.504064] usb usb4-port4: attempt power cycle
[    7.443613] usb 4-4: Device not responding to setup address.
[    7.663610] usb 4-4: Device not responding to setup address.
[    7.876041] usb 4-4: device not accepting address 8, error -71
[    8.111592] usb 4-4: Device not responding to setup address.
[    8.327605] usb 4-4: Device not responding to setup address.
[    8.540040] usb 4-4: device not accepting address 9, error -71
[    8.572064] usb usb4-port4: unable to enumerate USB device
[    9.857533] pcan: driver config [mod] [isa] [pci] [par] [usb] [net] 
[    9.864497] usbcore: registered new interface driver pcan
[    9.904208] usbcore: registered new interface driver btusb
[   10.031513] usb 4-3: Device not responding to setup address.
[   10.239510] usb 4-3: Device not responding to setup address.
[   10.444029] usb 4-3: device not accepting address 10, error -71
[... 1 lines cut ...]
[   10.883576] usb 4-3: Device not responding to setup address.
[   11.088053] usb 4-3: device not accepting address 11, error -71
[   11.120081] usb usb4-port3: attempt power cycle
[   12.055598] usb 4-3: Device not responding to setup address.
[   12.263581] usb 4-3: Device not responding to setup address.
[   12.468044] usb 4-3: device not accepting address 12, error -71
[   12.699564] usb 4-3: Device not responding to setup address.
[   12.907560] usb 4-3: Device not responding to setup address.
[   13.112043] usb 4-3: device not accepting address 13, error -71
[   13.144059] usb usb4-port3: unable to enumerate USB device
[   13.375576] usb 4-4: Device not responding to setup address.
[   13.583581] usb 4-4: Device not responding to setup address.
[... cut ...]
[   14.464062] usb usb4-port4: attempt power cycle
[   15.395571] usb 4-4: Device not responding to setup address.
[   15.603562] usb 4-4: Device not responding to setup address.
[   15.808017] usb 4-4: device not accepting address 16, error -71
[   16.039467] usb 4-4: Device not responding to setup address.
[   16.251491] usb 4-4: Device not responding to setup address.
[   16.456024] usb 4-4: device not accepting address 17, error -71
[   16.488053] usb usb4-port4: unable to enumerate USB device
```

elvyfeliz, 2019-03-05:
I have the same problem with the camera in the latest firmware. Everytime I contact someone from intel I get the same response, upgrade the firmware. Upgrading the firmware never works in this case.
@mikepurvis can you tell me more about the hardware solution you are implementing. Since, I have to do the same and look for a solution for this problem.

RealSenseCustomerSupport, 2019-04-23:
----------------------------------------------

The latest firmware 5.11.6.200 is now available via this link.
https://downloadcenter.intel.com/download/28724/Latest-Firmware-for-Intel-RealSense-D400-Product-Family
Please give it a try. Thank you

closed 2019-04-30 by RealSenseCustomerSupport

m9dfukc, 2019-06-01:
Hey, just wanted to report here ... I have the same issue and my cameras are running FW 5.11.6.200!
What a bummer ... makes this devices unusable in my case as unplugging and replugging is not a viable option in my case!

mikepurvis, 2019-06-03:
Yeah, we're still seeing it too. It's way less frequent, but still not fixed.

skribtsov, 2019-06-06:
Hi, Intel

We do like Intel Realsense - it's a great tool! However we need your help.
Our Intel Realsense D435 sensor has naturally 'died' yesterday. Before the death it showed up some degradation symptoms. 
- First of all on Linux it failed to report System Clock timestamps profile (although it did show it for the first week or so) and switched to Hardware Clock (which is not comfortable - we work with other sensors and with Hardware Clock profile we had to manually sync timestamps using complicated logic- that's why we switched to D435 from D415). We can't say what exactly caused it. Perhaps playing with sensor parameters in the setup tool from SDK and trying to load various json parameters file into the device lead to that. But can't say for sure. Intrestingly , on Windows10 it did show up System Clock profile. 
- Second - the video stream ceased to work during autorun when our application has been put to work as service (video worked only when manually run from console - hard to say... no such issue with D415. Perhaps something hasn't yet initialized in D435 during boot... very uncomfortable) 

Yesterday, trying to fix this issue and trying to use the "universal pill" - firmware update - something went wrong during firmware update and that's it. The device is not discoverable neither by applications nor by the firmware update too (DFU). Any suggestion, or we should send 'the body' to intel for autopsy?

best regards, Pavel

NatalieS1T2, 2019-08-14:
Hi @skribtsov @mikepurvis, we are also experiencing a similar issue. Did you do anything else beyond upgrading the firmware?

skribtsov, 2019-08-14:
@NatalieS1T2, Well the dead body was resurrected somehow using our admins, ...., yes they managed to force firmware update somehow. Nevertheless the ustable work of realsense is persistent even with new firmware. We also experienced strange behavior on updating the calibration parameters, which requires either sensor switch-off-and-on, or windows host reboot. So unstable work of realsense I can confirm for the current date, August 2019.

tispratik, 2019-10-28:
We are also facing this issue randomly. It was working well earlier, some days it just stops working without touching the cameras or upgrading/downgrading the firmware.

tispratik, 2019-11-01:
We found the issue. The issue is related to the specs of the USB cable. RealSense is rated at 2 Amps and most of the USB cables on the market are rated for 0.9 Amps. The moment we switched to a cable rated at 3 Amps, the issue went away. We have stress tested it with the new cables and we couldn't break the system, so far it's been robust. One more thing we did is to make sure to not have to use extension cables, so we have custom ordered 3 Amp USB C to USB A cables from Newnex for our needs.

Some good cables to look for: 
https://www.amazon.com/gp/product/B076CV7MPB
https://www.newnex.com/usb-type-c-cables-legacy.php (Look for power delivery 3 A)

Don't order general purpose cables like CableMatters or AmazonBasics, examples below:
https://www.amazon.com/gp/product/B00DMFB5OK
https://www.amazon.com/gp/product/B01GGKYR2O

eirikaso, 2020-04-03:
I have the same issue and it's not related to the cable as this is rated at 3A.

I did install a software to reset the usb connection and it seemed to help. Can anyone please try this solution and see if you are able to reset the device manually when the problem arise?
# install usb-reset
$ sudo snap install usb-reset

# Set up usb-reset
$ sudo snap connect usb-reset:hardware-observe core:hardware-observe
$ sudo snap connect usb-reset:raw-usb core:raw-usb

# reset realsense usb connection
$ usb-reset 8086:0ad4 
where 8086:0ad4 is usb device and manufacturer id. Can be found with "lsusb" in linux terminal

mkzein, 2020-11-20:
@eirikaso I have tried it. Did NOT work!

aruiplex, 2021-06-22:
I have the same issue on it. My camera is D435. AND I UPDATE TO THE LATEST FIRMWARE. I have to disconnect camera from usb socket everytime after running programe or reopen realsense-viewer. It is ridiculous and unstable. It really upset me.

**HOWEVER**, I use the original cable and the more stable usb sockets (which diectly connect on motherboard) on my PC, which solves my issue immediately 🙃.