realsenseai/librealsense #3335: RealSenseViewer Crash F200 on MacOS

hurfifi, 2019-02-27:
Required INFO
| Camera Model                       | F200
| Firmware Version                   | Last version available
| Operating System & Version |  MacOS  |                                 | 
| SDK Version                            |  { legacy / last }                          | 

Hello everyone,  ✋
I'm on macOS Sierra, I've installed the sdk via bush, I own a Lenovo Ideapad 500, with Intel F200 Camera built in. On the Mac, It's detected and appear as Intel F200 but it shows black screen in skype and other camera programs. After installing the SDK, I tried to open real sense viewer, it detects the cam but it crash and all of the others program don't detect the camera. 🤷‍♂️

Find below the crash code of real sense viewer : 

Process:               realsense-viewer [4306]
Path:                  /usr/local/Cellar/librealsense/2.18.0/bin/realsense-viewer
Identifier:            realsense-viewer
Version:               0
Code Type:             X86-64 (Native)
Parent Process:        ??? [4300]
Responsible:           realsense-viewer [4306]
User ID:               501

Date/Time:             2019-02-27 02:59:40.682 +0100
OS Version:            Mac OS X 10.12.6 (16G1815)
Report Version:        12
Anonymous UUID:        A8204A4F-37C6-6D68-65DE-7C2E0ED10776

Time Awake Since Boot: 2800 seconds

System Integrity Protection: disabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Segmentation fault: 11
Termination Reason:    Namespace SIGNAL, Code 0xb
Terminating Process:   exc handler [0]

VM Regions Near 0:
--> 
    __TEXT                 000000010068d000-00000001007e0000 [ 1356K] r-x/rwx SM=COW  /usr/local/Cellar/librealsense/2.18.0/bin/realsense-viewer

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   ???                           	000000000000000000 0 + 0
1   realsense-viewer              	0x000000010075e69c rs2::vao::vao(rs2::float3 const*, rs2::float2 const*, rs2::float3 const*, rs2::float3 const*, int, rs2::int3 const*, int) + 198
2   realsense-viewer              	0x000000010075e83b rs2::vao::create(rs2::obj_mesh const&) + 103
3   realsense-viewer              	0x0000000100759681 rs2::texture_visualizer::texture_visualizer(rs2::float2, rs2::float2) + 51
4   realsense-viewer              	0x000000010075962c rs2::visualizer_2d::visualizer_2d(std::__1::shared_ptr<rs2::texture_2d_shader>) + 28
5   realsense-viewer              	0x00000001007594fe std::__1::shared_ptr<rs2::visualizer_2d> std::__1::shared_ptr<rs2::visualizer_2d>::make_shared<std::__1::shared_ptr<rs2::splash_screen_shader> >(std::__1::shared_ptr<rs2::splash_screen_shader>&&) + 94
6   realsense-viewer              	0x0000000100741dde rs2::ux_window::open_window() + 2054
7   realsense-viewer              	0x000000010074295e rs2::ux_window::ux_window(char const*) + 530
8   realsense-viewer              	0x0000000100693b31 main + 95
9   libdyld.dylib                 	0x00007fffb1482235 start + 1

Thread 1:
0   libsystem_kernel.dylib        	0x00007fffb15b144e __workq_kernreturn + 10
1   libsystem_pthread.dylib       	0x00007fffb169b621 _pthread_wqthread + 1426
2   libsystem_pthread.dylib       	0x00007fffb169b07d start_wqthread + 13

Thread 2:
0   libsystem_kernel.dylib        	0x00007fffb15b144e __workq_kernreturn + 10
1   libsystem_pthread.dylib       	0x00007fffb169b48e _pthread_wqthread + 1023
2   libsystem_pthread.dylib       	0x00007fffb169b07d start_wqthread + 13

Thread 3:
0   libsystem_kernel.dylib        	0x00007fffb15b144e __workq_kernreturn + 10
1   libsystem_pthread.dylib       	0x00007fffb169b621 _pthread_wqthread + 1426
2   libsystem_pthread.dylib       	0x00007fffb169b07d start_wqthread + 13

[... 263 lines cut ...]
Stack                             10.0M        6 
VM_ALLOCATE                         48K        8 
__DATA                            22.2M      206 
__GLSLBUILTINS                    2588K        2 
__IMAGE                            528K        2 
__LINKEDIT                       119.6M        9 
__TEXT                           143.0M      208 
__UNICODE                          556K        2 
mapped file                       39.7M        8 
shared memory                     16.3M       10 
===========                     =======  ======= 
TOTAL                            507.4M      504 

Model: MacBookPro13,2, BootROM 251.0.0.0.0, 2 processors, Intel Core i5, 2,40 GHz, 4 GB, SMC 2.37f20
Graphics: Intel HD Graphics 520, Intel HD Graphics 520, Built-In
Memory Module: BANK 0/DIMM0, 4 GB, DDR3, 1600 MHz, Ramaxel Technology, RMT3170MN68F9F1600A
Bluetooth: Version 5.0.5f3, 3 services, 26 devices, 1 incoming serial ports
Network Service: Ethernet, Ethernet, en0
Serial ATA Device: ST1000LM024 HN-M101MBB, 1 TB
Serial ATA Device: PLDS    DVD-RW DA8A6SH
USB Device: USB 3.0 Bus
USB Device: Intel(R) RealSense(TM) 3D Camera (Front F200)
USB Device: Bluetooth HCI
USB Device: USB2.0-CRW
Thunderbolt Bus:

MartyG-RealSense, 2019-02-27:
What I interpret from your message is that you have a Lenovo Ideapad 500 and you have installed MacOS on it.  Is this correct, please?

If you have installed RealSense SDK 2.0 and are using its RealSense Viewer program, the F200 is not supported by that SDK.  For use with MacOS, your best option may be the older Librealsense SDK, now known as Legacy Librealsense.  This is compatible with F200.

https://github.com/IntelRealSense/librealsense/blob/v1.12.1/doc/installation_osx.md

hurfifi, 2019-02-27:
Affirm, I installed MacOs on it. I've done the guide but where do I launch the sdk ?

MartyG-RealSense, 2019-02-27:
I do not have direct experience of launching Librealsense 1.12 after it has been built (I do not work for Intel, and usually use pre-made binaries), so one of the Intel guys on this forum will be better able to answer this question.  I do apologize.

RealSenseCustomerSupport, 2019-03-01:
----------------------------------------------

@hurfifi  F200 is EOL for a long time. Suggest you change to the other supported RealSense cameras. Sorry for the inconvenience if any.

RealSenseCustomerSupport, 2019-03-04:
----------------------------------------------

@hurfifi  If no other questions, will close the ticket accordingly. Thanks!

closed 2019-03-11 by RealSenseCustomerSupport