realsenseai/librealsense #3336: Could I have a little tutorial on how to use the video feature? I want to display the video in a winforms picturebox if that’s possible ;)

jinxzzz, 2019-02-27:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {  D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Win10 | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C#}                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

dorodnic, 2019-02-27:
Hi @jinxzzz 
Did you look at [cs-tutorial-2-capture](https://github.com/IntelRealSense/librealsense/tree/master/wrappers/csharp/cs-tutorial-2-capture), I think it is more or less what you are looking for.

jinxzzz, 2019-02-27:
> Hi @jinxzzz
> Did you look at [cs-tutorial-2-capture](https://github.com/IntelRealSense/librealsense/tree/master/wrappers/csharp/cs-tutorial-2-capture), I think it is more or less what you are looking for.

Thanks for the quick response.
I wrote the winform program by referring to the source code you said.But the source code is written in WPF.WPF is still different from winform in many ways.The winform program I wrote according to this source code does not show depth images. Could you help me check my program and provide me with a small demo to show depth images in picturebox controls?Thank you very much!

Here is my code:

               Pipeline  pipeline;
               pipeline = new Pipeline();
               Colorizer colorizer;
               colorizer = new Colorizer();

               var cfg = new Config();
               cfg.EnableStream(Intel.RealSense.Stream.Depth, 640, 480, Format.Any, 30);
               cfg.EnableStream(Intel.RealSense.Stream.Color, 1280, 720, Format.Rgb8);
               var pp = pipeline.Start(cfg);
               
               var frames = pipeline.WaitForFrames();
               
               var colorFrame = frames.ColorFrame.DisposeWith(frames);
               var depthFrame = frames.DepthFrame.DisposeWith(frames);
               var p = pp.GetStream(Intel.RealSense.Stream.Depth) as VideoStreamProfile;
               var img = new Bitmap(p.Width, p.Height, PixelFormat.Format24bppRgb);
                pictureBox1.Image = img;

jinxzzz, 2019-03-06:
@RealSense-Customer-Engineering

ogoshen, 2019-04-01:
Problem with @jinxzzz's code isn't Winform\WPF related, it's that depth images aren't in a format compatible for display. The default depth format is `Format.Z16`, which is 16-bit linear depth in device units.
You'll need to convert this for something that can be displayed, this is what the `Colorizer` processing block is for, it returns a color-mapped frame with `Format.Rgb8`.

RealSenseCustomerSupport, 2019-04-09:
----------------------------------------------

Hi jinxzzz,

Given what ogoshen mentioned, wonder if you have any update? Or do you still need help?

Thanks!

RealSenseCustomerSupport, 2019-04-18:
----------------------------------------------

Hi jinxzzz,

If there is nothing else, this one will be closed soon.

Thanks!

closed 2019-04-26 by RealSenseCustomerSupport