realsenseai/librealsense #3338: Aligning color to depth (again)

AndrewFielding, 2019-02-27:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {  D415} | 
| Firmware Version                   | (05.11.01.00) | 
| Operating System & Version |   {Win (10) /   | 
| Kernel Version (Linux Only)    |                                     | 
| Platform                                 | PC/  |
| SDK Version                            |  { 2.18.1}                          | 
| Language                            |  {matlab }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description

I am trying to use the align.m function to align video and depth streams using the MATLAB wrapper. I am using the code chunk suggested in issue #2455.

align_to = realsense.stream.color;
alignedFs = realsense.align(align_to);
.....
 fs = pipe.wait_for_frames();
 aligned_frames = alignedFs.process(fs);

When using this I get the following error message corresponding to the line of script.

alignedFs = realsense.align(align_to);  

Can anyone help with this error message (see below)?

Error using realsense.align
Error: File: align.m Line: 9 Column: 35
A constructor call to superclass realsense.filter appears after the object is used, or after a return.

Error in acquire_color_depth (line 45)
aligned = realsense.align(align_to);

RealSenseCustomerSupport, 2019-03-21:
----------------------------------------------

Hi AndrewFielding,

Besides the changes mentioned in #2455, you will need to do another change in align constructor.
Together 2 changes:
a) For https://github.com/IntelRealSense/librealsense/blob/master/wrappers/matlab/align.m#L8
Replace "this.objectHandle" with "out", which is used in next line for fitler().
b) As mentioned in #2455, comment out validateattributes() in process()
https://github.com/IntelRealSense/librealsense/blob/master/wrappers/matlab/align.m#L17

Thanks!

RealSenseCustomerSupport, 2019-03-22:
----------------------------------------------

Hi AndrewFielding,

Wonder if you have any update?

Thanks!

AndrewFielding, 2019-03-26:
Hi,

The fix appears to work., at least it seems to run ok and the colour and
depth appear to be aligned. Will this change appea rin a future update?

many thanks
Andrew

lramati, 2019-03-26:
Pull request #3577 addresses this, and has already been merged into
development.

closed 2019-04-02 by RealSenseCustomerSupport