realsenseai/librealsense #3341: Segmentation Faults Because of Small RAM Size???

ShadowHunter0510, 2019-02-27:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Operating System & Version | Ubuntu 16 | 
| Platform                                 | Raspberry Pi |

### Issue Description
It appears that a few other users and I have been encountering segmentation faults on the Raspberry Pi. I believe that this may be because there is not enough dedicated RAM size. My program relies on an infinite loop, which would of course increase the chances of segmentation faults, since it tracks distance in real time. Perhaps is there any way to “clear” the data that would be obtained from each frame captured when the program loops again? The program may be storing this data that is only necessary for relaying the distance for one instance at a time.

dorodnic, 2019-02-27:
From my experience, boards like this often come without configured swap file. Perhaps this a better option? 
Overall the memory footprint of the library is not that big, but it was also never designed for low-memory situations.

ShadowHunter0510, 2019-02-27:
Hi @dorodnic 

I already have 2GB of swap file size. Should I increase this? Or is there an alternative solution?

dorodnic, 2019-02-28:
It should be plenty. Are you sure the issue is related to out-of-memory situation? What data are you seeing using htop?

ShadowHunter0510, 2019-02-28:
@dorodnic 

After using htop, I noticed that the memory used just hovers around 450 MB. It jumps to around maximum capacity right when the program starts and periodically increases with every frame captured. I’m not very sure what is causing this issue.
[image]
[image]

dorodnic, 2019-02-28:
Are you getting stable memory when running `rs-capture` and other examples? 
There was issue of memory leak with python that was localised to a leak in numpy (#3164), perhaps this can be related?

ShadowHunter0510, 2019-03-01:
Hi @dorodnic 

I actually can’t open rs-capture or any other example. Notably, rs-capture opens up a window for a few seconds then promptly crashes after.

dorodnic, 2019-03-02:
I think it should be fine. I'm just suggesting one possible source for the problem. Perhaps the application is doing some additional heavy processing, maybe storing multiple frames?

RealSenseCustomerSupport, 2019-03-07:
----------------------------------------------

@ShadowHunter0510 Any update for Dorodnic's question? Thanks!

jdwolk, 2019-03-09:
I'm also getting segfaults for, i.e. `rs-depth`, `rs-color`, etc, on Beaglebone Black. I have a 2G swap file allocated also.

My specs:
* Beaglebone Black
* Ubuntu 16.04
* Kernel 4.4
* D435, firmware 05.11.01.100
* librealsense 2.18.1 (also got the same thing on 2.19.0)

In my case, I'm also getting `"Frame didn't arrived within 5000"` before the segfault of most of these tools.

`realsense-viewer` launches for me (I'm using lxde as a lightweight desktop) + recognizes the camera over USB 2.0 but I can't flip the "switches" on the left.

RealSenseCustomerSupport, 2019-03-22:
----------------------------------------------

@jdwolk  Did you try if you still get segmentation fault with rs-depth if change to lower resolution or FPS? Such as 480x270@30fps or 640x480@15fps

RealSenseCustomerSupport, 2019-03-26:
----------------------------------------------

@jdwolk Any update? Thanks!

jdwolk, 2019-03-29:
We ended up moving to Raspberry Pi which opened up separate issues (see https://github.com/intel-ros/realsense/issues/652, https://github.com/intel-ros/realsense/issues/669 ). I can try again in a few days with the Beaglebone.

RealSenseCustomerSupport, 2019-04-26:
----------------------------------------------

@jdwolk Do you still need support on this issue? Please update. Thanks!

filipematosinov, 2019-04-26:
I am having similar behaviour with the Raspberry Pi. Getting segmentation fault using the realsense-viewer or python scripts using pyrealsense2.

RealSenseCustomerSupport, 2019-04-28:
----------------------------------------------

@filipematosinov Did you try rs-depth to see if it works or not?

RealSenseCustomerSupport, 2019-05-05:
----------------------------------------------

@filipematosinov @jdwolk @ShadowHunter0510 Any other questions for this ticket? Please update. Thanks!

filipematosinov, 2019-05-06:
I have tried rs-depth having the same issue

RealSenseCustomerSupport, 2019-05-07:
----------------------------------------------

@filipematosinov What's the resolution and FPS configuration when you get issue for rs-depth?

RealSenseCustomerSupport, 2019-05-13:
----------------------------------------------

@filipematosinov Looking forward to your reply? Could you please update? Thanks!

RealSenseCustomerSupport, 2019-05-21:
----------------------------------------------

@filipematosinov If no other questions, will close the ticket at this point. Thanks!

closed 2019-05-28 by RealSenseCustomerSupport