realsenseai/librealsense #3342: Newer firmware requires plug-cycle after reboot to be detected

jkflying, 2019-02-27:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D435  | 
| Firmware Version                   | 5.10.6 and newer (5.10.3 unaffected) | 
| Operating System & Version |   Ubuntu 16.04  | 
| Kernel Version (Linux Only)    |  4.9.56-intel-pk-standard                                        | 
| Platform                                 | Up Board |
| Language                            |  ROS/C++                         | 
| Segment			|  Robot                          | 

### Issue Description
We are attempting to upgrade our systems from 5.9.13 firmware to the latest version to eliminate DSO-10002 and auto-exposure issues, but are hitting a bug:

On boot, the realsense is not detected until it is unplugged and plugged in again. This seems to have been introduced between releases 5.10.3 and 5.10.6. This has been verified multiple times by flashing the firmware forward and backwards and rebooting. Note that the device shows up before the plug-cycle using `lsusb`, but the ROS node doesn't publish any information (camera info, depth cloud, anything), despite the ROS node being re-launched many times.

In all cases, unplugging and plugging in the camera after boot solves the issue for firmware versions 5.10.6 and above. Note that this is not a viable workaround for our embedded system.

I'm happy to test intermediate development firmware versions to help locate where this was introduced, or to provide any other system logs requested.

MartyG-RealSense, 2019-02-27:
Placing a hardware_reset instruction near the start of a script will help to avoid having to perform USB unplug-replug.  An example script is in the link below.

https://github.com/IntelRealSense/librealsense/issues/2161#issuecomment-408779150

jkflying, 2019-02-28:
That's a pretty ugly workaround for a firmware regression, it increases the boot time significantly (I saw people adding a 10 second delay following the link you provided) and requires us to maintain our own branch of the ROS node. 

For now we will stay on firmware version 5.10.3, since it works.

RealSenseCustomerSupport, 2019-03-02:
----------------------------------------------

Hi

I just tried D435 with FW5.10.6 on "UP Board" under Ubunto 16.04, but couldn't reproduce the issue on either USB2.0 ports or the OTG port. I can always see the camera after lsusb, 8086:0b07., no matter cold boot or warm boot.

I also tried it on NUC with Ubunto 16.04, neither could I see the issue.

Did I miss something?

jkflying, 2019-03-02:
`lsusb` shows the device, however trying to start a ROS node gives an empty stream.

I haven't tried the realsense-viewer since my kernel doesn't have display drivers compiled in, and x-forwarding over SSH doesn't work with opengl.

RealSenseCustomerSupport, 2019-03-08:
----------------------------------------------

Hi jkflying,

Would you please try a few things:
1. Use rs-enumerate-device tool to see if you can list out the device:
rs-enumerate-devices -s

2. Use rs-save-to-disk to test if you can just save to depth+RGB to local disk, instead of stream out.
rs-save-to-disk

If both work well, it would be something with ROS. And we will then take from there.

And at the end can you help clarify the ROS wrapper and RealSense SDK version info?

Thanks!

RealSenseCustomerSupport, 2019-03-14:
----------------------------------------------

Hi jkflying,

Wonder if you have any updates?

Thanks!

jkflying, 2019-03-15:
Hi, sorry for the delays.

`rs-enumerate-devices` looks good, I get output for stereo module (IR and depth) and RGB Camera.
```
# rs-enumerate-devices                                                                       
 Device info:                                                                                                             
    Name                          :     Intel RealSense D435                                                              
    Serial Number                 :     817612070347                                                                      
    Firmware Version              :     05.10.06.00                                                                       
    Recommended Firmware Version  :     05.09.14.00                                                                       
    Physical Port                 :     /sys/devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2:1.0/video4linux/video0          
    Debug Op Code                 :     15                                                                                
    Advanced Mode                 :     YES                                                                               
    Product Id                    :     0B07                                                                              
    Usb Type Descriptor           :     2.1                                                                               
                                                                                                                          
Stream Profiles supported by Stereo Module                                                                                
 Supported modes:                                      stream     resolution fps   format                                 
    Infrared 1    1280x720      @ 6Hz   Y8                                                                                
    Infrared 1    1280x720      @ 4Hz   Y8                                                                                
    Infrared 1    640x480       @ 30Hz  Y8                                                                                
    Infrared 1    640x480       @ 15Hz  Y8                                                                                
    Infrared 1    640x480       @ 6Hz   Y8                                                                                
    Infrared 1    640x480       @ 4Hz   Y8                                                                                
    Infrared 1    480x270       @ 60Hz  Y8                                                                                
    Infrared 1    480x270       @ 30Hz  Y8                                                                                
    Infrared 1    480x270       @ 15Hz  Y8                                                                                
    Infrared 1    480x270       @ 6Hz   Y8                                                                                
    Infrared 1    480x270       @ 4Hz   Y8                                                                                
    Depth         1280x720      @ 6Hz   Z16                                                                               
    Depth         1280x720      @ 4Hz   Z16                                                                               
    Depth         640x480       @ 30Hz  Z16                                                                               
    Depth         640x480       @ 15Hz  Z16                                                                               
    Depth         640x480       @ 6Hz   Z16                                                                               
    Depth         640x480       @ 4Hz   Z16                                                                               
    Depth         480x270       @ 60Hz  Z16                                                                               
    Depth         480x270       @ 30Hz  Z16                                                                               
    Depth         480x270       @ 15Hz  Z16                                                                               
    Depth         480x270       @ 6Hz   Z16                                                                               
    Depth         480x270       @ 4Hz   Z16

Stream Profiles supported by RGB Camera
 Supported modes:                                      stream     resolution fps   format   
    Color         1280x720      @ 6Hz   RGB8
    Color         1280x720      @ 6Hz   Y16
    Color         1280x720      @ 6Hz   BGRA8
    Color         1280x720      @ 6Hz   RGBA8
    Color         1280x720      @ 6Hz   BGR8
    Color         1280x720      @ 6Hz   YUYV
    Color         640x480       @ 30Hz  RGB8
    Color         640x480       @ 30Hz  Y16
    Color         640x480       @ 30Hz  BGRA8
    Color         640x480       @ 30Hz  RGBA8
    Color         640x480       @ 30Hz  BGR8
    Color         640x480       @ 30Hz  YUYV
    Color         640x480       @ 15Hz  RGB8
    Color         640x480       @ 15Hz  Y16
    Color         640x480       @ 15Hz  BGRA8
    Color         640x480       @ 15Hz  RGBA8
    Color         640x480       @ 15Hz  BGR8
    Color         640x480       @ 15Hz  YUYV
    Color         640x480       @ 6Hz   RGB8
    Color         640x480       @ 6Hz   Y16
[... cut ...]
Here are the log files from the failed launch with 5.10.6 firmware:
[roslaunch.zip](https://github.com/IntelRealSense/librealsense/files/2970222/roslaunch.zip)

Here are the logs from a successful launch with 5.10.6, where the realsense was unplugged and plugged again after a cold boot:
[success.zip](https://github.com/IntelRealSense/librealsense/files/2970237/success.zip)

ROS wrapper version is 2.1.0, Realsense SDK is 2.16.0 (taken from /usr/lib/x86_64-linux-gnu/cmake/realsense2/realsense2Config.cmake)

Thanks for your help!

RealSenseCustomerSupport, 2019-03-15:
----------------------------------------------

Hi jkflying,

Thanks for the update. We will try to reproduce it here.

Meanwhile, wonder if you can try the latest FW 5.11.4 and RS SDK 2.19.1 (or 2.19.0) + ROS 2.2.1 combination?

Thanks!

jkflying, 2019-03-15:
Hello

Firmware 5.11.4 has the same issue.

I've tried to update the Realsense/ROS but it seems some of the setting names have changed, so I'll need a bit more time to modify our launch files.

RealSenseCustomerSupport, 2019-03-21:
----------------------------------------------

Hi jkflying,

Have you got some update on launch files? Wonder if ROS 2.2.1 has same issue off your UP board? If so, does it happen with every launch file that ROS has? Or only with your modified versions?

Thanks!

RealSenseCustomerSupport, 2019-04-09:
----------------------------------------------

Hi jkflying,

Wonder if there is any update? Do do you still need help on this one?

Thanks!

jkflying, 2019-04-09:
Hi, sorry for the long delay.

We have scheduled in an update to latest librealsense and ROS wrapper with the parameter changes required. I will let you know once this is done, if we still have the issue with latest firmware.

RealSenseCustomerSupport, 2019-04-23:
----------------------------------------------

Hi jkflying,

Any update? Or do you guys still need help on this one?
If there is nothing else, we will close this one soon.

Thanks!

closed 2019-05-02 by RealSenseCustomerSupport